boltgolt/howdy #318: Gstreamer warnings

ajnart, 2020-02-28:
_Please describe the issue in as much detail as possible, including any errors and traces_

While attempting to use howdy, it gives me these warnings:
```
[ WARN:0] global /build/opencv/src/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```

----

Linux distribution (if applicable): Arch Linux 86_64

Howdy version: 2.5.1-2

meeas, 2020-04-10:
Thank you!  After a recent update on Debian Testing, I started experiencing this same issue.  Confirming your patch fixed it for me.

Cottand, 2020-04-17:
Can confirm the same

principis, 2020-06-24:
This has to be changed again for 2.6.0
see https://github.com/boltgolt/howdy/blob/master/src/recorders/video_capture.py#L116

closed 2021-04-02 by boltgolt

baelter, 2021-06-10:
Fix got rid of the WARNs, but INFO logs can still create issues for GUI applications, eg. 1Password.
Tried setting `OPENCV_LOG_LEVEL=ERROR` in `/etc/environment` but still get info logs that causes an infinite loop of retries.'

Seeing this (repeated until the app is killed) in the logs:
```
Jun 10 13:36:13 fedora polkit-agent-helper-1[3851]: [HOWDY] Attempting facial authentication for user anders
Jun 10 13:36:14 fedora gnome-shell[1963]: Unknown line '[ INFO:0] global /builddir/build/BUILD/opencv-4.5.2/modules/videoio/src/videoio_registry.cpp >
```

petre2dor, 2021-07-30:
@baelter  were you able to fix this? 
I would like to enable face recognition for polkit on fedora 34 with gnome and this issue is a blocker.

baelter, 2021-07-30:
@petre2dor Nope 😔

citizenserious, 2024-07-01:
I am seeing the same error on fedora 40 today. Howdy is working, but it takes a lot of time for it to start.

ysz0507, 2025-04-08:
A small pointer from my side: I got the same warning on manjaro. I could resolve the problem by applying this commit from the current beta branch: https://github.com/boltgolt/howdy/commit/b04ffe5bd83683949db53bcaf2b91559e30d8e4c