boltgolt/howdy #323: IR emitter not turning on on Razer Blade 15 advanced (mid 2019 edition)

Kelvin858, 2020-03-05:
Howdy installed fine and was working well until I rebooted into windows and used windows hello, which made it so that now that I've booted back into manjaro, I can't even manually turn on the IR emitter. Howdy has no errors and `howdy test` starts the video window and recieves input from the IR camera, but it is almost completely black (minus some IR from flourecent lights or tv remotes etc). TLDR, the camera is sending input but not actually able to see anything because the emitter isn't blinking or even on. Not sure if it's a device/driver issue or a howdy issue with windows hello, but it definitely has something to do with that service, as it only breaks when hello is used in between reboots.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch (manjaro)

fschoenfeldt, 2020-03-15:
Having a similar issue on Ubuntu and Blade Advanced

TheButlah, 2020-06-22:
Having this same issue on the Razer Blade 2020 Advanced. I've confirmed that the infra red camera works, but the image is just too dark for it to make out frames most of the time

Here is an example of the IR output from ffplay:
[image]

Here is `howdy test`:
[image]

That seems quite dark to me. I'm noticing as well that usually when I do windows hello, a red light rapidly flashes, which presumably is the IR emitter and would illuminate my face, but this doesn't happen on linux

Note that `/dev/video2` appears to be the IR Camera, and `/dev/video0` is the RGB.
Output of `v4l2-ctl -d /dev/video0 --list-ctrls`:
```v4l2-ctl -d /dev/video0 --list-ctrls
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-180 max=180 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=90 max=150 step=1 default=120 value=120
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=7 step=1 default=3 value=3
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
              exposure_absolute 0x009a0902 (int)    : min=2 max=1250 step=1 default=156 value=156 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
```
The same command but for `/dev/video2` returns nothing.

<details><summary>Here is `lsusb -v`</summary>

```

Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.04
  iManufacturer           3 
  iProduct                2 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
[... 1505 lines cut ...]
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
```
</details>

TheButlah, 2020-06-22:
Also note that I am aware of `dark_threshold`. But I think that is more addressing a symptom than the cause - if someone can figure out how to turn on the IR emitter, that is the best solution. Also `dark_threshold` still isn't enough for most things in my case

nalin29, 2020-07-25:
Actually I have the same issue. When booting from windows it stops working but if you enter the bios and then boot to linux this fixes the issue. I am not sure why this the case but on my razer blade this reproducable. It is only windows that causes this issue not other linux distributions not sure why.

ppriest, 2020-08-30:
See https://github.com/PetePriority/chicony-ir-toggle/blob/master/main.c and https://github.com/boltgolt/howdy/issues/269.

I found that thread before this - the small program to enable the IR blaster works in the RazorBlade 15 Advanced (mid-2019) at least with the default device/unit/selector. Just build/install it and add it to say /etc/rc.local.
However, I found that it is still a bit too dark when illuminating and you need to set the dark_threshold pretty high. Playing around with the source I managed to get it much brighter, but haven't completely figured it out yet.

closed 2021-04-02 by boltgolt

KindRoach, 2021-04-25:
@nalin29 Follow your steps and it works again. Unbelievable!

j4ger, 2022-03-31:
> See https://github.com/PetePriority/chicony-ir-toggle/blob/master/main.c and https://github.com/boltgolt/howdy/issues/269.
>
> I found that thread before this - the small program to enable the IR [...]

Would you please share your modifications please 'cause I'm currently having the same issues and have no clue at all. Any form of help would be great, thanks!