boltgolt/howdy #324: Install failing

mchassy, 2020-03-05:
Here are the final lines of the `sudo apt install howdy`
```
Setting up howdy (2.5.1) ...

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.7-3) ...
Processing triggers for libc-bin (2.30-0ubuntu2.1) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
I've searched for similar issues already, and my issue has not been reported yet.
Linux distribution (if applicable): Pop!_OS 19.10

Howdy version: 2.5.1

I have looked at the other issues, and I have not found anything which points to a solution.
I have also googled the errors, but `TypeError: 'module' object is not callable` is pretty generic.

srueegger, 2020-03-11:
I can not Confirm the Issue. I have installed howdy (over the PPA) on PopOS 19.10 and it works finde.

boltgolt, 2020-03-11:
Please search for other issues with the same error trace, this issue has been reported already and a fix is in #265

closed 2020-03-11 by boltgolt