boltgolt/howdy #327: Does Howdy *need* IR sensors?

adrfantini, 2020-03-28:
Hi and thanks for this project, it looks very interesting and cool!
I'd like to try it out, but I do not have a Windows Hello laptop - so no IR sensors. Are they really needed, or are they just added security?

The project description says `Use your built-in IR emitters and camera in combination with facial recognition to prove who you are.`, does this mean that IR sensors are necessary for this to work reliably?

Doomsdayrs, 2020-04-01:
Not at all, I have it installed on laptops without IR sensors. 
It seems the models will just be more based on image then mapping of depth and such. 

TLDR; No, But you will be less secure, and some issues with recognition

boltgolt, 2020-04-01:
Yes, there is not depth mapping as there really isn't a way to generate depth data. So a normal camera will do fine, but it's a LOT more easier to fool with a picture of you for instance

adrfantini, 2020-04-06:
Got it, thanks! Maybe could be a bit more clear in the README

closed 2020-04-06 by adrfantini

boltgolt, 2020-04-06:
Don't want to encourage people to use Howdy with normal webcams though, not very secure

adrfantini, 2020-04-06:
True, even though in some environments could be OKish (e.g. house computer with kids around).
Anyway a big disclaimer would be enough imho.