boltgolt/howdy #328: Can't add face model

micahlt, 2020-03-29:
Hi!  I'm trying to add my face to Howdy, but I'm getting an error with `howdy add`.  Here's the traceback log: 

```bash
File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.2.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```

Any help is appreciated!  :+1:

boltgolt, 2020-04-01:
Please search for that error code first, it's a common issue

micahlt, 2020-04-01:
All right, will do.

micahlt, 2020-04-06:
@boltgolt After a long while Googling, it seems like it must be an error with the source code.  I poked around a little, but I don't know enough Python to understand the code.  I don't have an IR emitter, is it possible that's what is generating the error?

boltgolt, 2021-04-02:
It is not

closed 2021-04-02 by boltgolt