boltgolt/howdy #329: Howdy doesn't install on Fedora 32

kevintigelaar, 2020-04-04:
_Please describe the issue in as much detail as possible, including any errors and traces_
When trying to install Howdy on Fedora 32 I am getting this error. I've done research but cannot find any solutions online. 

```
Problem: package howdy-2.5.1-2.fc31.x86_64 requires python3dist(v4l2), but none of the providers can be installed
  - conflicting requests
  - nothing provides python(abi) = 3.7 needed by python3-v4l2-0.2-1.fc31.noarch
(try to add '--skip-broken' to skip uninstallable packages)
```

Anyone has an idea how to fix this?

----

I've searched for similar issues already, and my issue has not been reported yet. Yes

Linux distribution (if applicable): Fedora 32

Howdy version:

boltgolt, 2020-04-24:
Looks like python3 itself can't be found? Very weird.
I'm not that used to fedora unfortunately, and i don't have any other reports of this. Is rpm itself configured okay?

persello, 2020-04-29:
Happens the same to me, on a fresh install of Fedora 32. I followed the instructions on the README file.

principis, 2020-05-02:
I was tired of the broken dependencies so I built it myself. Anyone interested, you can find it [https://mirror.principis.be/fedora/claque/releases/32/x86_64/Packages/](here). 
I also changed the dependencies because a lot where unnecessary. Only python3-pam, python3-dlib and python3-opencv are necessary. 

(You probably shouldn't depend on that repo, because it's for personal use.)

denzilferreira, 2020-05-02:
> I was tired of the broken dependencies so I built it myself. Anyone interested, you can find it [here](https://mirror.principis.be/fedora/claque/releases/32/x86_64/Packages/).
> I also changed the dependencies because a lot where unnecessary. Only pam_python, python3-dlib and python3-opencv are necessary.
> 
> (You probably shouldn't depend on that repo, because it's for personal use.)

Awesome, just installed and seems to work. Maybe a PR of the changes? :)

principis, 2020-05-02:
> Awesome, just installed and seems to work. Maybe a PR of the changes? :)

Sure but the changes are pretty minor and @boltgolt seems to be inactive as he hasn't approved #319 yet.
Anyway, the only problem is that the dependencies aren't compiled for fedora 32, which is what I did.

principis, 2020-05-03:
@kevintigelaar @persello @denzilferreira I've created a [copr repo](https://copr.fedorainfracloud.org/coprs/principis/howdy/) with al the required packages, seemed better than my own repo.

lverissimo, 2020-05-03:
@principis thank you so much! I was able to install, config, test and add users. however I was not able to get it to work. It seems to be missing the following folder: /lib64/security/howdy

principis, 2020-05-04:
> @principis thank you so much! I was able to install, config, test and add users. however I was not able to get it to work. I am running on gnome on X11. How should I configure pam.d file?

I just installed it on my desktop (which doesn't have a camera, so I can't fully test it), but it seems to work. Can you try it for `sudo`?

Just at the following line to the top of `/etc/pam.d/sudo`
```
auth       sufficient   pam_python.so /lib64/security/howdy/pam.py
```

There is a small guide at the readme of the [repo](https://copr.fedorainfracloud.org/coprs/principis/howdy/)

For authentication in gnome, you can try [this](https://github.com/williamwlk/my-red-corner/blob/master/README_PAM.txt). I use KDE so I can't test it. 
If you want to login using howdy, (and if you are using sddm) you can find the guide on my [repo](https://copr.fedorainfracloud.org/coprs/principis/howdy/).

lverissimo, 2020-05-04:
@principis I have gone over those and this is what I get. I have debug mode turned on, and I have removed the opencv workaround you mentioned in another post.

[sudo] password for xxxxx: 
Attempting face detection
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/videoio_registry.cpp (187) VideoBackendRegistry VIDEOIO: Enabled backends(7, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for FFMPEG
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (690) open OpenCV | GStreamer: /dev/video0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (723) open OpenCV | GStreamer: mode - FILE
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (INTEL_MFX): glob is 'libopencv_videoio_intel_mfx*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for INTEL_MFX
Face detection timeout reached
Attempting face detection
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/videoio_registry.cpp (187) VideoBackendRegistry VIDEOIO: Enabled backends(7, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for FFMPEG
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (690) open OpenCV | GStreamer: /dev/video0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (723) open OpenCV | GStreamer: mode - FILE
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (INTEL_MFX): glob is 'libopencv_videoio_intel_mfx*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for INTEL_MFX
Face detection timeout reached
Attempting face detection
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/videoio_registry.cpp (187) VideoBackendRegistry VIDEOIO: Enabled backends(7, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for FFMPEG
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (690) open OpenCV | GStreamer: /dev/video0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (723) open OpenCV | GStreamer: mode - FILE

[... cut ...]
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (INTEL_MFX): glob is 'libopencv_videoio_intel_mfx*.so', 1 location(s)

(python3:778539): GStreamer-CRITICAL **: 16:04:26.247: gst_element_post_message: assertion 'GST_IS_ELEMENT (element)' failed
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for INTEL_MFX
Face detection timeout reached

principis, 2020-05-04:
> @principis I have gone over those and this is what I get. I have debug mode turned on, and I have removed the opencv workaround you mentioned in another post.

@lverissimo I can't remember which workaround you're referring to.

Anyway I think it works, but your face doesn't get recognized. Can you show me your /etc/pam.d/sudo config file? It's strange that it keeps looping

lverissimo, 2020-05-04:
@principis I am getting there! 

I was able to get a working configuration with two changes:
1. I removed my glasses and added a photo without glasses. (howdy is geek biased!)
2. moved the  auth statement to the beggining of the /etc/pam.d/sudo script. I had it at the end. 

 [root@lverissimo pam.d] # more /etc/pam.d/sudo 
 #%PAM-1.0
 auth       sufficient   pam_python.so /lib64/security/howdy/pam.py
 auth       include      system-auth
 account    include      system-auth
 password   include      system-auth
 session    optional     pam_keyinit.so revoke
 session    required     pam_limits.so
 session    include      system-auth
# originally was here
# auth       sufficient   pam_python.so /lib64/security/howdy/pam.py

now I just need to get a glasses image of me that works!! 

thank you so much for your support!!!

lverissimo, 2020-05-04:
Sorry... not over yet. Only sudo-i works. 

I changed gdm-password and system-auth.
sudo -i works like a charm.
su - attempts to face recognize but IR is not led. Even with selinux on permissive mode.
lock screen attempts facial recognition, IR is led up, but times out. I am without glasses!

I gues now I have something messed up with my selinux configuration, I applied what you suggested:
ausearch -c 'python3' --raw | audit2allow -M my-python3
semodule -i my-python3.pp 

Thanks in advance.

principis, 2020-05-05:
> For selinux I applied what you suggested, but I am not sure if it can be related.
> ausearch -c 'python3' --raw | audit2allow -M my-python3
> semodule -i my-python3.pp

@lverissimo 
This was necessary for me to make login using sddm work. I didn't completely looked in to it, if there was an other way to solve it, because I'm quite busy.

> su - attempts to face recognize but IR is not led. Even with selinux on permissive mode.
I just tried adding howdy to `/etc/pam.d/su` and it can't find any face models. 

That is normal because you are logging in to a different user so the user passed to howdy is the one you called su with.

I always add multiple pictures, some with, some without headphones, different positions and different hairstyles (I'm always wearing glasses) (which is important because my webcam is on the bottom...) and that works perfectly! :)

lverissimo, 2020-05-05:
@principis Thank you so much for the help. You were super! I never got it working so far. It is really a great usability improvement. Thank you for your time.

omarnadhem, 2020-07-17:
I have the same problem 
pip install face-recognition
Collecting face-recognition
  Using cached https://files.pythonhosted.org/packages/1e/95/f6c9330f54ab07bfa032bf3715c12455a381083125d8880c43cbe76bb3d0/face_recognition-1.3.0-py2.py3-none-any.whl
Requirement already satisfied: Pillow in /usr/lib64/python3.8/site-packages (from face-recognition) (7.0.0)
Requirement already satisfied: numpy in ./.local/lib/python3.8/site-packages (from face-recognition) (1.19.0)
Requirement already satisfied: face-recognition-models>=0.3.0 in ./.local/lib/python3.8/site-packages (from face-recognition) (0.3.0)
Collecting dlib>=19.7
  Using cached https://files.pythonhosted.org/packages/f1/40/e407366816646bb452578997e6ca2140f004eec8f611e8512ee429ded2be/dlib-19.20.0.tar.gz
Requirement already satisfied: Click>=6.0 in ./.local/lib/python3.8/site-packages (from face-recognition) (7.1.2)
Installing collected packages: dlib, face-recognition
    Running setup.py install for dlib ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6b23a2zt/dlib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6b23a2zt/dlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-r1cb7dww/install-record.txt --single-version-externally-managed --compile
         cwd: /tmp/pip-install-6b23a2zt/dlib/
    Complete output (45 lines):
    running install
    running build
    running build_py
    package init file 'dlib/__init__.py' not found (or not a regular file)
    running build_ext
    Building extension for Python 3.8.3 (default, May 29 2020, 00:00:00)
    Invoking CMake setup: 'cmake /tmp/pip-install-6b23a2zt/dlib/tools/python -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-install-6b23a2zt/dlib/build/lib.linux-x86_64-3.8 -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=Release'
    CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
    CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
    CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
    -- Configuring incomplete, errors occurred!
    See also "/tmp/pip-install-6b23a2zt/dlib/build/temp.linux-x86_64-3.8/CMakeFiles/CMakeOutput.log".
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-6b23a2zt/dlib/setup.py", line 223, in <module>
        setup(
      File "/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib64/python3.8/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib64/python3.8/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/usr/lib64/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3.8/site-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib64/python3.8/distutils/command/install.py", line 556, in run
        self.run_command('build')
      File "/usr/lib64/python3.8/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib64/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib64/python3.8/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/usr/lib64/python3.8/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib64/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-6b23a2zt/dlib/setup.py", line 135, in run
        self.build_extension(ext)
      File "/tmp/pip-install-6b23a2zt/dlib/setup.py", line 172, in build_extension
        subprocess.check_call(cmake_setup, cwd=build_folder)
      File "/usr/lib64/python3.8/subprocess.py", line 364, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-install-6b23a2zt/dlib/tools/python', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-install-6b23a2zt/dlib/build/lib.linux-x86_64-3.8', '-DPYTHON_EXECUTABLE=/usr/bin/python3', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6b23a2zt/dlib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6b23a2zt/dlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-r1cb7dww/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.

closed 2026-11-16 by boltgolt