boltgolt/howdy #330: dlib install and illegal instructions

pawan-xd, 2020-04-07:
_Please describe the issue in as much detail as possible, including any errors and traces_

(system - Manjaro KDE latest)
(newbie here so i might make a few mistakes)

hello, so i have been trying to install howdy. first i tried to install it from pamac (i have AUR support turned on) . When pamac is installing, it also installs/builds dependencies. The problem seems to be occuring when 'python-dlib' is getting built. After about 70% progress is where the problem starts, the building process starts to loop for some reason (see pastebin below). When i tried to install manually using the pkg build from AUR, i get this

pawan@manjaro howdy]$ makepkg -si
==> Making package: howdy 2.5.1-2 (Tuesday 07 April 2020 05:40:53 PM)
==> Checking runtime dependencies...
==> Installing missing dependencies...
[sudo] password for pawan: 
error: target not found: python-dlib
==> ERROR: 'pacman' failed to install missing dependencies.
==> Missing dependencies:
  -> python-dlib
==> Checking buildtime dependencies...
==> ERROR: Could not resolve all dependencies.

So i tried to manually install 'python-dlib' from the AUR PKGBUILD. The same looping thing starts to happen that happened when pamac tried to build 'python-dlib'.

I may not be able to perfectly describe the problem so i have pasted the output from the terminal in a pastebin, here https://pastebin.com/sARrWduy .

In the pastebin i have aborted the install where the looping starts to happen because if i let the looping continues, my pc starts get unresponsive. The install does get finished (successfully?) but it takes about 20 mins. 
howdy gets succesfully installed but when i try to do
>>sudo howdy add
i get the output
>>Illegal instructions

i have also tried using the '--user' argument to see if it would do anything but it doesnt . 
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version:

boltgolt, 2020-04-24:
This is some very deep Arch stuff that i don't know much about, sorry. Hoping someone here is more knowledgeable of Arch/Manjaro

lololololkl, 2020-10-13:
I've met the same error.But don't know how to do.

closed 2021-04-02 by boltgolt