boltgolt/howdy #331: IR camera don't regconize face

k41r4, 2020-04-10:
Everything works fine with webcam but when i changed to IR camera, it doesn't regconize my face, `sudo howdy add` give "No face detected, aborting" and 'sudo howdy test' show some blurry image with no circle 
[image]

I'm using Linux Mint and the lastest version of howdy (or i thing so, just istalled it today)

and313, 2020-04-13:
> Everything works fine with webcam but when i changed to IR camera, it doesn't regconize my face, `sudo howdy add` give "No face detected, aborting" and 'sudo howdy test' show some blurry image with no circle, and the red led on IR camera doesn't blink as it does with window hello
> [image]
> 
> I'm using Linux Mint and the lastest version of howdy (or i thing so, just istalled it today)

what happens if you use the command ffplay /dev/video2 (try different numbers tll you find the ir output)? do you get a visible image there?

k41r4, 2020-04-13:
> what happens if you use the command ffplay /dev/video2 (try different numbers tll you find the ir output)? do you get a visible image there?

I tried to change it to my webcam and it work just fine, but when it's come to IR camera it just like that.
By the way, i got some problem and have to reinstall the whole system and everything works just fine

boltgolt, 2020-04-24:
Looks like a classic issue, i think the white noise is very similar to #147

closed 2020-04-24 by boltgolt