boltgolt/howdy #332: Missing dependencies

wrtechnopath, 2020-04-12:
Tried installing and got this error, installed pam python using `pip install python-pam`, rebooted and still the same error.

> I'm using openSUSE` Tumbleweed 20200409

> Error and image to follow:

[conflicts.txt](https://github.com/boltgolt/howdy/files/4466300/conflicts.txt)
[image]

boltgolt, 2020-04-24:
Could you try `pip3 install python-pam`?

pguinet, 2020-05-08:
Installed wheel and python-pam but still the same error (nothing provides pam-python).
Changed alternatives in order to have Python 3 as default.

NAME="openSUSE Tumbleweed"
# VERSION="20200502"

# python --version
Python 3.8.2

# pip3 install python-pam
Requirement already satisfied: python-pam in /usr/local/lib/python3.8/site-packages (1.8.4)

I even copied pam.py and python_pam-1.8.4.dist-info/ to /usr/lib/python3.8/site-packages

closed 2021-04-02 by boltgolt