boltgolt/howdy #333: howdy does not respect $EDITOR

NuclearDog, 2020-04-13:
howdy (2.5.1) does not respect the standard unix $EDITOR environment variable.

In `src/cli/config.py`, should replace:

```
# Default to the nano editor
editor = "/bin/nano"
```

With something more like:
```
editor = os.environ.get("EDITOR", "nano")
```

(That's also switched from using `/bin/nano` -> `nano`. This will respect the user's $PATH variable as far as locating the nano binary.)

boltgolt, 2020-04-14:
Hmm, but howdy does fetch that exact variable a few lines below that? It only uses nano if no `/etc/alternatives/editor` or `$EDITOR` is found

NuclearDog, 2020-04-14:
Ah sorry about that. I saw EDITOR wasn't doing anything, saw the "editor = '/bin/nano'" line, and just kinda stopped reading there. Will you accept lack of coffee as an excuse? :disappointed: 

Looks like my issue is more the prioritization -- I'd argue that preferring `/etc/alternatives/editor` over `$EDITOR` is backwards and that `if`/`elif` should be reversed. `/etc/alternatives/editor` is set on the system level by the administrator, while `$EDITOR` is supposed to be the per-user override.

The current behaviour is definitely opposite any other application I've run into, as apparently my `/etc/alternatives/editor` is linked to nano and this is the first time I've noticed because I have `$EDITOR='vim'` set and that's been respected by e.g., visudo, etc.

boltgolt, 2020-04-24:
Hm, okay. Fixing in next release

boltgolt, 2020-06-22:
Implemented exactly as suggested

closed 2020-06-22 by boltgolt

xz-dev, 2020-08-20:
The bug is back.

[image]

arifer612, 2020-11-08:
@xz-dev The problem at hand seems to be due to the fact that you are elevated when running `sudo howdy config`. Your user-defined `EDITOR` is no longer relevant. A temporary fix you can do to handle this would be to change the symlink for your root default editor to `nvim`.

    sudo ln -sf /etc/alternatives/editor /path/to/nvim

@boltgolt I think the closure of this issue should be reconsidered.

boltgolt, 2020-11-08:
@arifer612 what would a better solution be in your opinion? I think it's hard to make this work everywhere as everyone expects

arifer612, 2020-11-08:
@boltgolt I apologise for sounding very direct, I did not mean for that.

I was searching for a way to do so and I think I found a method that might work for this use-case. Hooking the `$user` name into the following:

    editor = subprocess.run(["su", "-", "$user", "-c" "source ~/.profile; echo $EDITOR"], stdout=subprocess.PIPE).stdout.decode()

It gave me the right `$EDITOR` when executed as root. I have yet to see if it can be fit `config.py` without any issues yet, but I can look into with your permission.