boltgolt/howdy #335: Use howdy exclusively for sudo (and similar)

brunoalmeida-fairtiq, 2020-04-14:
This identification system is quite interesting but it appears to be active for all identification means automatically.
I want to change that. I want to limit howdy for `sudo` (and potentially other similar such as `su`) commands.
What can I configure? What should I change?

Additionally, just to make sure we are in agreement about this: I don't want pam_mount to ask me for password.
I followed this [answer's advice](https://unix.stackexchange.com/a/273894)
Adding `disable_interactive` worked.

boltgolt, 2020-04-24:
I've made a new wiki entry just for this, [check it out](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types)

closed 2020-04-24 by boltgolt

brunoalmeida-fairtiq, 2020-04-24:
Thank you. It's done.