boltgolt/howdy #336: Howdy doesn't start on lockscreen/login GDM

avanishsubbiah, 2020-04-15:
_Please describe the issue in as much detail as possible, including any errors and traces_
Testing with sudo howdy works fine. I have a Surface Laptop 3 so there's a white light that shows up whenever the IR and Camera are activated, so it is easy to tell when they are not being used. With sudo this lights up and everything works; however, on lock and login screen the light doesn't turn on, which means that the cameras are not activated. I saw a similar-looking issue around Kubuntu but I don't have any unusual disk encryption, and I'm not using SDDM.

tail /var/log/auth.log doesn't work because auth.log doesn't exist there, so I'm not sure where to start to find out what's going wrong.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch Linux x86_64

Howdy version: I'm not sure I'm using the AUR howdy

avanishsubbiah, 2020-04-15:
Actually, did some more testing and it seems like it doesn't work also with any gnome authentication prompts. Ex. unlocking certain settings, opening grub customizer, installing software from software center. It only starts up the camera in the terminal.

avanishsubbiah, 2020-04-15:
Nvm, I figured it out. I needed to change where I added the auth in pam.d

closed 2020-04-15 by avanishsubbiah

ruffson, 2020-07-29:
Hey there, I have the same problem, could you please explain what fixed it for you?

ruffson, 2020-07-29:
Ok nevermind, I figured it out:
You have to add the `PAM` rule to `/etc/pam.d/gdm-password` if you wanna unlock or login if you are using GDM/GNOME.