boltgolt/howdy #337: Howdy working for sudo authentication but not for login

haji-ali, 2020-04-15:
I am getting an `Unknown error: 1` when I try to login with howdy. However authenticating sudo works as expected. Looking into `/var/log/auth.log`, I see the following log

```
Apr 15 09:21:22 thinkpad /lib/security/howdy/pam.py[27348]: Traceback (most recent call last):
Apr 15 09:21:22 thinkpad /lib/security/howdy/pam.py[27348]:   File "/lib/security/howdy/pam.py", line 68, in pam_sm_authenticate
Apr 15 09:21:22 thinkpad /lib/security/howdy/pam.py[27348]:     return doAuth(pamh)
Apr 15 09:21:22 thinkpad /lib/security/howdy/pam.py[27348]:   File "/lib/security/howdy/pam.py", line 62, in doAuth
Apr 15 09:21:22 thinkpad /lib/security/howdy/pam.py[27348]:     pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Unknown error: " + str(status)))
Apr 15 09:21:22 thinkpad /lib/security/howdy/pam.py[27348]: PamException: Conversation error
```

Any idea how to resolve this?
I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu

Howdy version: 2.5.1

closed 2020-04-19 by haji-ali

ramiel, 2020-06-04:
I have the same error. Why is this closed? Did you solve it?

ramiel, 2020-06-04:
Ok, I solved following #243