boltgolt/howdy #338: Error when installing and Workaround

TheFibonacciEffect, 2020-04-15:
During the installation I got the following Error:
```

c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/dlib_python.dir/src/image.cpp.o] Error 4
CMakeFiles/dlib_python.dir/build.make:326: recipe for target 'CMakeFiles/dlib_python.dir/src/image.cpp.o' failed
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/dlib_python.dir/all] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/dlib_python.dir/all' failed
make: *** [all] Error 2
Makefile:83: recipe for target 'all' failed
Traceback (most recent call last):
  File "setup.py", line 257, in <module>
    'Topic :: Software Development',
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 129, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 24, in run
    self.build()
  File "/usr/lib/python3.6/distutils/command/install_lib.py", line 109, in build
    self.run_command('build_ext')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 133, in run
    self.build_extension(ext)
  File "setup.py", line 173, in build_extension
    subprocess.check_call(cmake_build, cwd=build_folder)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
```

When trying to run `sudo howdy add` I got the following Error: 
```
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 18, in <module>
    import dlib
ModuleNotFoundError: No module named 'dlib'
Unknown error: 1
[sudo] password for caspar: 
No module named 'dlib'

Can't import the dlib module, check the output of
pip3 show dlib
```

Installing delib using `$ sudo -H pip3 install dlib` seems to fix the issue foe me.

Linux distribution (if applicable): Debian/Ubuntu

Howdy version: Nov 19, 2019

boltgolt, 2020-04-24:
Dlib installation is such a pain during install, will need to be handled differently in the next release

boltgolt, 2020-06-22:
No real way yet to reliablyinstall dlib without building from source yet, bumping to next release

boltgolt, 2020-12-03:
Closing this now as there have been no other dlib compilation errors posted and as there does not seem to be a better solution right now

closed 2020-12-03 by boltgolt