boltgolt/howdy #341: X Error: BadDrawable (invalid Pixmap or Window parameter) 9

choyrim, 2020-04-24:
_Please describe the issue in as much detail as possible, including any errors and traces_

When running `sudo howdy test`, I see an infinite loop of the following error on the console after the window opens:
```
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x1a00010
```
----

I've searched for similar issues already, and my issue has not been reported yet.
I searched for "baddrawable" and "pixmap" in issues and found no other issue.

Linux distribution (if applicable):
```
$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.4 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.4 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

Howdy version:
```
$ sudo apt list howdy
Listing... Done
howdy/bionic,bionic,now 2.5.1 all [installed]

```

choyrim, 2020-04-24:
googling the error i found a similar error at
https://github.com/P0cL4bs/WiFi-Pumpkin-deprecated/issues/53

and provides a workaround
https://github.com/P0cL4bs/WiFi-Pumpkin-deprecated/issues/53#issuecomment-222999004

so I was able to get it to run with
```
sudo QT_X11_NO_MITSHM=1 howdy test
```

boltgolt, 2020-04-24:
See also #23

closed 2020-04-24 by boltgolt