boltgolt/howdy #342: [Feature Request] Continious retraining

Doomsdayrs, 2020-04-29:
Humans often change overtime. Hair style changes, clothing changes, bruises and cuts appear and skin wrinkles. 

The issue i have had with howdy is that my beard throws the detection off.

Is it possible that, if it detects a well matching face, that howdy will remake the model with the addition of the new data. So that howdy will be able to grow a "life cycle" of your face, such as in my case the growth and removal of my beard

boltgolt, 2020-04-29:
Sounds like a really good idea, especially if it's done in the background after authentication!

Doomsdayrs, 2020-04-30:
thought it would come with a few requirements

1. A secure way to save the data
2. A way to remove wrong models (IE someone who looks close to you, but isn't you)
3. (Optional) Way to train the AI to reject individuals that do not match, with a reporting mechanism to tell the user who attempted to login.

thomasesr, 2021-02-25:
The option to automatically train the model when you correctly type the password so that the model gets better each time you put your password in would be a nice feature.

JeffersonTan, 2024-08-16:
I think this is a great feature, since I use Howdy on my laptop and whenever I bring my laptop to another place with a completely different background, it fails to detect my face. I think this in theory could help fix this issue I'm having.