boltgolt/howdy #343: Howdy on Ubuntu 20.04 stops working - IR camera ok

loekaas, 2020-04-29:
_Please describe the issue in as much detail as possible, including any errors and traces_
Using Dell XPS 13 9300. On a fresh install of Ubuntu, Howdy work perfectly. But after a very short time, it stops working. Sometimes it starts to work again after reboot but the last couple of reboots it does not work. The IR cameras seem to work, but no frames are captured.
Authentication just times out. Trying to add new model hangs forever, pressing ctrl c give me this output:

oystein@XPS13-9300:[~]$ sudo howdy add
Face detection timeout reached
[sudo] password for oystein: 
NOTICE: Each additional model slows down the face recognition engine slightly
Press Ctrl+C to cancel

Adding face model for the user oystein
Enter a label for this new model [Model #6] (max 24 characters): ff

Please look straight into the camera
^CTraceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 146, in <module>
    ret, frame = video_capture.read()
KeyboardInterrupt
 
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): 20.04

Howdy version:

Liemaeu, 2020-05-08:
Same problem.
Ubuntu 20.04

sudo howdy -U richard add
Adding face model for the user richard
Enter a label for this new model [Initial model] (max 24 characters): Face 
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 112, in <module>
    video_capture = cv2.VideoCapture(config.get("video", "device_path"))
AttributeError: module 'cv2' has no attribute 'VideoCapture'

Worked perfectly on Ubuntu 19.10 (before I upgraded to 20.04)

sylvaticus, 2020-05-11:
Same (similar) problem here.
All fine in 18.04, now in 20.04 when installed, it asks if the light on the camera went on, but they never.. after a bit I replied YES, but of course it is not working...

Antiprogrammer-user, 2020-06-01:
Hi all, I ran into a similar issue. I have a 2019 hp specter x360 diamond cut. I installed ubuntu 20.04 alongside Windows 10 and I did all the installation steps but it cant add a face and randomly throws an error message. 

File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.2.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

ppwaskie, 2020-06-14:
I'm having the same issue, Dell XPS 9300 13".  If I register new face models, then unlocking the lock screen or sudo authentication works fine via the face recognition.  But after rebooting or returning from suspend, the camera turns on, but nothing seems to happen, and the face recognition times out.

I have Ubuntu 20.04 installed, and have 8 face models stored now for the same user, from different angles, etc.  I've tried clearing all the models and adding them back, but everything seems to fail to recognize once a reboot or resume happens.

TheButlah, 2020-06-20:
Similar issue here: Hangs at the same point, in my case I have not yet been able to successfully add a face model due to the bug.

ruixunbao, 2020-06-21:
Ubuntu 20.04 no longer support python 2 while howdy build upon python 2.

ppwaskie, 2020-06-21:
This seems to happen mostly when my Dell XPS 9300 comes out of Sleep (pm-suspend).  The camera and IR detectors fire up, but the timeout is reached.  I can sometimes fix that with a full reboot, or I reinstall howdy completely.

Nothing suspect in the auth.log or other logs that I can find.  I'm happy to provide any debug info if requested.

boltgolt, 2020-06-21:
I see all kinds of different unrelated issues in this thread:

---

@loekaas @ppwaskie @TheButlah Seems like you have a lot of dark frames, could you try setting the `dark threshold` to 100 in the config?

---

@Liemaeu You have a borked or old OpenCV install, try reinstalling cv2 for python3

---

@sylvaticus You seem to not have an compatible IR camera sorry

---

@Antiprogrammer-user This is a VERY common issue, could you please search for existing issues next time, see #160

---

Locking this thread now to prevent even more unrelated issues being reported here

closed 2020-06-21 by boltgolt