boltgolt/howdy #344: 99% (Installation on Ubuntu stays at 99% most of the time because of the C build stuff)

ElSenor64, 2020-04-29:
_Please describe the issue in as much detail as possible, including any errors and traces_
I am running Ubuntu Focal Fossa (20.04 LTS) and during installation via ppa repository, I noticed the progress bar got to 99% pretty quickly and stayed there for over half of the installation. I couldn't help but be reminded of the days when I used Windows; it does the same thing pretty much always. 

I also noticed it was caused by some C build (cpython?), and got _lots_ of python deprecation warnings. I tried to save all of them, but there were so many that they exceeded the scroll-back history. I did manage to get this screenshot as a sample.

I know this is a minor issue that can be solved later, but I thought I'd just post it here. Deprecations aren't pretty.
I do love this Idea, and even though I haven't tried it out in all aspects yet, I've been impressed by the quality and care that's gone into this. I really think it could ship in Ubuntu by default soon. Thanks!
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu Focal Fossa (20.04 LTS)

Howdy version: 2.5.1

boltgolt, 2020-04-29:
So there are 2 issues here:

The "loading bar stuck at 99%" issue stems from the fact that the Howdy post installation script is executed by APT after basically everything else, but apt does not know how long this script will run for. This script configures the camera, does the PAM setup, installs dlib (more on that later), and much more. Some of this takes quite a while, but for APT this only seems like 1 step. I don't believe there is a way to make APT aware how long post installation takes.

As for the python errors: These come from dlib, a library Howdy uses to match face models. We might need to move to a newer version of dlib. I think we're using version 19 right now

ElSenor64, 2020-04-30:
Sorry about that. For some reason I didn't separate the two issues in my mind. Should I close or delete this one and create two separate issues or just leave this one as is?

boltgolt, 2020-04-30:
It's fine in 1 issue for now, i think there's only 1 fixable issue here and that's the dlib stuff

ElSenor64, 2020-04-30:
Okay, thanks!

kobozo, 2020-07-30:
I have the same problem, installed today on ubuntu 20.04

closed 2026-11-16 by boltgolt