boltgolt/howdy #345: Multiple Cameras

ElSenor64, 2020-04-30:
I often plug my laptop into a sort of docking station setup with an external webcam (which I use most of the time), but there is the occasion when I want to use Howdy authentication when my laptop is unplugged. (And use the builtin webcam.) I don't know how easy it would be to implement, but it would be nice for there to be support for multiple webcam devices.
Thanks again for this wonderful software!

boltgolt, 2020-06-21:
Does sound nice, if more people are interested i'll consider implementing it

Barltarman, 2020-07-13:
I have the exact same scenario, would love to have this functionality. The only issue I would have though is that the external webcam doesn't have any IR emitters so would be reliant upon the IR coming from the laptop. Would that even be possible? Love the app btw, so incredibly useful!

dav-is, 2020-11-13:
This is something I'd be interested in. Ideally, it would first try to use my external webcam, and if it wasn't detected it would fallback to the webcam on my laptop. My external webcam has IR emitters, as well as my laptop.

tdaniely, 2020-11-24:
To make it work for now.

1. Add file `/lib/udev/rules.d/99-howdy-camera-select.rules`
```
KERNEL=="video2", SYMLINK+="videohowdy"
KERNEL=="video0", SYMLINK+="videohowdy"
```
Replace video0 / video2 with your devices.
The first one is the camera from the dock.
The second one is the builtin.

2. Refresh udev
`sudo udevadm trigger`

3. Configure howdy to the new device.
``
```
sudo howdy config
device_path = /dev/videohowdy
# if you close your lid when docked then also
ignore_closed_lid = false
```

4. `sudo howdy add` with both cameras seperately.

5. magic.

Worked for me on Ubuntu 20.10 on dell 5511 & Logitech BRIO.

boltgolt, 2021-04-02:
See #515

closed 2021-04-02 by boltgolt

kabeersvohra, 2021-11-24:
> To make it work for now.
> 
>     1. Add file `/lib/udev/rules.d/99-howdy-camera-select.rules`
>        Replace video0 / video2 with your devices. The first one is the camera from the dock. The second one is the builtin. The first device in the list that is found will grab the symlink, You can chain as many as you want.
> 
> 
> ```
> KERNEL=="video2", SYMLINK+="videohowdy"
> KERNEL=="video0", SYMLINK+="videohowdy"
> ```
> 
>     2. trigger udev `sudo udevadm trigger`
> 
>     3. `sudo howdy config`
>        `device_path = /dev/videohowdy`
>        
>        1. if you close your lid when docked then also
>           `ignore_closed_lid = false`
> 
>     4. `sudo howdy add` with both cameras seperately.
> 
>     5. magic.
> 
> 
> Worked for me on Ubuntu 20.10 on latitude 5511 with Logitech BRIO through WD19TB dock.
> 
> Ain't linux great xD

@tdaniely this doesn't seem to be working for me, I think I may have got my udev rules wrong. I have `dev/video2` as my laptop camera and `dev/video4` for my docked camera. I know that this works because when I manually change the howdy config to each camera, they each individually get used. I tried to set this as my `/lib/udev/rules.d/99-howdy-camera-select.rules` but it always seems to try the laptop camera:
```
KERNEL=="video4", SYMLINK+="videohowdy"
KERNEL=="video2", SYMLINK+="videohowdy"
```
I also tried it in the other order and it just uses the laptop camera, I know that the udev rules are being triggered because when it set it to just
```
KERNEL=="video4", SYMLINK+="videohowdy"
```
it works fine. Any idea why it is not choosing priority correctly based upon which camera is available?

Tiger862000, 2022-05-03:
Hey, can you please link some of the solutions in the wiki section! That would be very helpful. Thank you.

BloodDrag0n, 2022-10-21:
I have a doubt, does howdy can use both video feeds from the laptop's integrated webcam and intergrated IR cam to perform more accurately. I don't know whether it exits already in howdy or not.

boltgolt, 2022-10-31:
See #718

rosydawn6, 2024-11-20:
> To make it work for now.
>
>     1. Add file `/lib/udev/rules.d/99-howdy-camera-select.rules`
>        Replace video0 / video2 with your devices. The first one is the camera from the dock. The second one [...]

I had neglected to change the parameter ignore_closed_lid from false to true. Thanks