boltgolt/howdy #348: IR camera doesn't work (regular camera does)

stautonico, 2020-05-13:
_Please describe the issue in as much detail as possible, including any errors and traces_
I keep trying to add a model using my IR camera on my laptop but I keep getting cv2.error: !_src.empty() in function "cvtColor". I selected my regular camera (/dev/video0) and I have no problems. I can add a model and login successfully. I try to switch my camera to my IR camera (/dev/video2) and I get the error. I can't register the model. The light on the IR camera does turn on and I know that the IR camera works because Windows Hello works properly (dual boot). Here's my full stacktrace:

```
elliot@popos:~$ sudo howdy add
Adding face model for the user elliot
Enter a label for this new model [Model #2] (max 24 characters): 
[ WARN:0] global /io/opencv/modules/videoio/src/cap_v4l.cpp (998) tryIoctl VIDEOIO(V4L2:/dev/video2): select() timeout.

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.2.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor

```

Linux distribution (if applicable):
Pop_OS! (Ubuntu/Debian derivative)

Howdy version:
2.5.1 (apt)

StormFalcon32, 2020-05-24:
I'm having a similar error

boltgolt, 2020-06-21:
This just means that you're camera is sending invalid frames to Howdy, see the bigger thread #160

closed 2020-06-21 by boltgolt