boltgolt/howdy #353: Any way to disable INFO logging from opencv?

ardevd, 2020-05-16:
Currently using Howdy as sudo authentication and it's a bit annoying that verbose output from opencv is being posted to the terminal every time.

```
sudo ls
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/videoio_registry.cpp (187) VideoBackendRegistry VIDEOIO: Enabled backends(7, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for FFMPEG
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (690) open OpenCV | GStreamer: /dev/video0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (723) open OpenCV | GStreamer: mode - FILE
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (INTEL_MFX): glob is 'libopencv_videoio_intel_mfx*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for INTEL_MFX
Identified face as user
```

What's the best way to disable this behaviour?

lverissimo, 2020-06-04:
I am having the same issue.

boltgolt, 2020-06-21:
There's currently no way to do this in python unfortunately, we'll have to wait for OpenCV to implement this upstream

closed 2020-06-21 by boltgolt

dmaglio, 2021-02-07:
set envroiment variable as `export OPENCV_LOG_LEVEL=OFF` to hide messages

expl0r3rgu1, 2022-07-19:
> set envroiment variable as `export OPENCV_LOG_LEVEL=OFF` to hide messages

This did the job for me