boltgolt/howdy #354: Howdy using RGB camera instead of IR

theLMGN, 2020-05-17:
I have a generic USB Windows Hello camera, and whenever I use Howdy add on `/dev/video0`, the LED for the RGB camera turns on, and Howdy adds a model. Trying to actually use this model or test it fails. Trying to use it on `/dev/video1` fails to add.

```
thelmgn@vulpes:~$ v4l2-ctl -d /dev/video0 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : RGB-IR Camera: RGB-IR Camera
	Bus info         : usb-0000:02:00.0-1.4
	Driver version   : 5.4.30
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV' (YUYV 4:2:2)
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 614400
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=50
                       contrast 0x00980901 (int)    : min=0 max=100 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=32 value=32
                            hue 0x00980903 (int)    : min=-4 max=4 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=1 max=16 step=1 default=8 value=8
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=5000 value=5000 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=20 step=1 default=4 value=4
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
thelmgn@vulpes:~$ v4l2-ctl -d /dev/video1 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : RGB-IR Camera: RGB-IR Camera
	Bus info         : usb-0000:02:00.0-1.4
	Driver version   : 5.4.30
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04a00000
		Metadata Capture
		Streaming
		Extended Pix Format
Priority: 2
Format Metadata Capture:
	Sample Format   : 'UVCH' (UVC Payload Header Metadata)
	Buffer Size     : 1024
```

----

Linux distribution (if applicable): Pop!_OS 20.04 LTS

Howdy version: 2.5.1

boltgolt, 2020-06-21:
What does the output of `sudo howdy test` look like for the IR camera?

theLMGN, 2020-06-21:
I don't currently have access to Linux right now, but if I remembered it just crashed with an error message.

boltgolt, 2020-06-21:
I would really like to see that error

closed 2021-04-02 by boltgolt