boltgolt/howdy #356: Howdy on Flatpak or Snap

qiyoulu, 2020-05-22:
So the story began when I was trying to get Howdy to work on Fedora Silverblue. It installed fine with layering, but I just couldn't make configurations, since it's a read-only filesystem.
Long story short, is there any difficulties to making a Flatpak or Snap package of Howdy? It would make it available to a lot more users who do not have access to the Ubuntu PPA or the AUR. I've seen feature requests on here wanting Howdy on Solus, Void, Gentoo, Clear Linux, etc. As a Fedora user, I've also had to switch 3 COPRs because of dependency issues.
Thank you all!

boltgolt, 2020-06-21:
I currently have no plans to make one, but i welcome anyone who wants to try their hand in making one

fatkaratekid, 2021-10-03:
I will try creating a snap package, let's see how it goes.

calvinweb, 2022-08-14:
Howdy is work as a pam module so it is inpossible to bring it to snap/flatpak

whitty, 2022-10-03:
CLI programs published by snaps can be "executed" by any user (maybe an asterisk or two here) - it might be possible to snap 95% - and just have to provide the pam hook externally - I'm going to have to have a play.

I'm here because I panicked when I saw the PPA was install pip packages _as root_ (this is a way to end up with a broken base-system).  snaps would provide a better solution to that issue.

Perhaps there is another solution to that issue that doesn't involve snaps though,....

Aqua1ung, 2023-03-08:
Let me add my voice to those clamoring for a _Clear Linux_ port ... or, even better, a Flatpak! As far as I know, the Flatpak framework is somewhat more flexible than Snap, as there is [a way to adjust app permissions](https://flathub.org/apps/details/com.github.tchx84.Flatseal), which may not exist in Snap.

encrustace, 2024-01-15:
Any update/progress?