boltgolt/howdy #357: kernel 5.6.14 - Assertion ELFW(R_TYPE) (reloc->r_info) == R_X86_64_RELATIVE' failed! Unknown error: 127

diepes, 2020-05-24:
**_Please describe the issue in as much detail as possible, including any errors and traces_**
Upgraded my Ubuntu kernel to 5.6.16
Now get a error, cant quite pinpoint it but when i disable howdy its gone.

$ sudo ls
Inconsistency detected by ld.so: ../sysdeps/x86_64/dl-machine.h: 540: elf_machine_rela_relative: Assertion `ELFW(R_TYPE) (reloc->r_info) == R_X86_64_RELATIVE' failed!
Unknown error: 127
Inconsistency detected by ld.so: ../sysdeps/x86_64/dl-machine.h: 540: elf_machine_rela_relative: Assertion `ELFW(R_TYPE) (reloc->r_info) == R_X86_64_RELATIVE' failed!
Unknown error: 127
[sudo] password for diepes:

$ sudo howdy disable true
sodo work as always with password, and when i re-enable howdy i get the error above , and then the password login.

----

**I've searched for similar issues already, and my issue has not been reported yet.**

**Linux distribution (if applicable):**
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

$ uname -a
Linux diepes-HP 5.6.14-050614-generic #202005200733 SMP Wed May 20 07:38:05 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

**Howdy version:**
2.5.1

boltgolt, 2020-06-21:
Weird, looks like it's an actual kernel bug. It's out of scope for Howdy but thanks for reporting, hope you can find help elsewhere or a kernel update fix

closed 2020-06-21 by boltgolt

diepes, 2020-06-26:
I tried again today.
howdy is already the newest version (2.6.0).
Linux  5.6.14-050614-generic #202005200733

Works great, no errors.

Loongphy, 2023-11-30:
If anyone met this problem, reboot is the best solution.