boltgolt/howdy #360: Error when reading from camera at random

Woople74, 2020-05-29:
_Please describe the issue in as much detail as possible, including any errors and traces_
It seems that howdy randomly fails to read from my IR sensor
here is the error 

[ WARN:0] global /io/opencv/modules/videoio/src/cap_v4l.cpp (998) tryIoctl VIDEOIO(V4L2:/dev/v4l/by-path/pci-0000:04:00.4-usb-0:1:1.0-video-index0): select() timeout.

Unknown camera, please check your 'device_path' config value.

Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 193, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.2.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

Unknown error: 1

It does work perfectly the rest of the time.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 20.04

Howdy version: 2.5.1

boltgolt, 2020-06-21:
Do you have any other applications open that also use the camera when Howdy fails?

Woople74, 2020-10-09:
No it does it just after booting up on a fresh ubuntu install

closed 2026-11-16 by boltgolt