boltgolt/howdy #362: Different camera used for IR and camera

galook, 2020-06-02:
Hello,
I was installing Howdy on Ubuntu 20.04 via apt, and it asked me if the IR blasters are running during the installation. I have gone through the four options (devices), the first two didn't do anything, the third one fired up the IR blasters and the camera led was turned off, the fourth (last) one only fired up the camera led, not the IR.

What can I do to use both the camera and the IR blasters, how I suppose face recognition is intended to work?

boltgolt, 2020-06-04:
What does running `ls /dev/v4l/by-path` return?

galook, 2020-06-04:
It does return this:
```
pci-0000:00:14.0-usb-0:5:1.0-video-index0
pci-0000:00:14.0-usb-0:5:1.0-video-index1
pci-0000:00:14.0-usb-0:5:1.2-video-index0
pci-0000:00:14.0-usb-0:5:1.2-video-index1
```
The ones ending -index0 don't do anything on installation, the first one ending -index1 are the IR blasters, the second one is the camera.

thecreativeone91, 2020-06-05:
I have the same issue on Ubuntu 20.04 with my lenovo Carbon X1. I can get it working using the webcam only but it does not use IR. Funny thing is Zoom Meetings shows both cameras and the IR one turns on the IR lights and has a B&W image. I can use Howdy using my normal webcam but not the IR one it won't detect a face on those. 

This is what mine returns

```
pci-0000:00:14.0-usb-0:5:1.0-video-index0
pci-0000:00:14.0-usb-0:5:1.0-video-index1
pci-0000:00:14.0-usb-0:8:1.0-video-index0
pci-0000:00:14.0-usb-0:8:1.0-video-index1
```

boltgolt, 2020-06-21:
Could be a `dark_threshold` issue, try upping that to 100 in the config

closed 2021-04-02 by boltgolt