boltgolt/howdy #363: Build from source

LeoSavard, 2020-06-03:
Hi,
I've been using Howdy with my old macbook pro on pop_os since it works even with a non-ir camera. However I've switched my laptop to clear Linux to benefit from the better optimizations regarding speed and overall responsiveness on my 8 year old system. Is it possible to build Howdy from source or do you require that packages be distro-specific?

boltgolt, 2020-06-04:
Well, Howdy is written in python 3, so it should really run anywhere. There's no real build process and you basically only have to configure Howdy and point PAM in the right direction. The [openSUSE wiki](https://twitter.com/NicolasEstrup/status/1268570956672520198) has a good explanation of what you have to do

qiyoulu, 2020-06-06:
that's not the opensuse wiki

boltgolt, 2020-06-06:
Mispaste sorry!

LeoSavard, 2020-06-07:
> Well, Howdy is written in python 3, so it should really run anywhere. There's no real build process and you basically only have to configure Howdy and point PAM in the right direction. The [openSUSE [...]

Thank you! It's basically plug and play!

closed 2020-06-07 by LeoSavard