boltgolt/howdy #364: Use howdy to unlock 1Password

Naragorn, 2020-06-09:
How cool would it be to use howdy to unlock third party apps like 1Password!

1Password has this feature for Apple and Windows however not for Linux.

boltgolt, 2020-06-21:
While this would be really cool, i think it's outside the scope of Howdy. It's up to 1Password to fix this

closed 2020-06-21 by boltgolt

FIUSHerson, 2021-09-13:
> How cool would it be to use howdy to unlock third party apps like 1Password!

I haven't tested it yet, but 1Password should be able to use Howdy if you enable System Authentication in 1Password.

Naragorn, 2021-09-16:
You are correct! This works, I just tested it.

To do this enable System Authentication like so:
https://support.1password.com/system-authentication-linux/

Then add this line
`auth sufficient pam_python.so /lib/security/howdy/pam.py`

to your 
`etc/pam.d/system-auth`

boltgolt, 2021-09-16:
That's really cool to hear!