boltgolt/howdy #365: Configurable timeout to read from camera

VanVan, 2020-06-09:
It should be good if we could configure the timeout to read from the camera.
There are times where camera doesn't work with an error message, but the timeout is too long, and we know that if it doesn't work in a second, it will never works, so  if we can configure timeout, it can be less boring to wait a long timeout before typing our password. :)

boltgolt, 2020-06-21:
There is a `timeout` config option that should do exactly what you're describing. Unfortunately, due to a type it's currently broken but will be fixed in the next release

closed 2020-06-21 by boltgolt

VanVan, 2020-06-21:
I can indeed confirm that the default timeout of 4seconds in config.ini did not apply and that the delay is much longer.
Thank you for your confirmation to put it back in the next release 👍