boltgolt/howdy #367: Dell XPS 9500 IR camera not recognized

ctrlbrk42, 2020-06-13:
Ubuntu 20.04 LTS
Dell XPS 15 9500 (2020 model)

Camera light turns on, indicating image recording -- but the IR sensors do not turn on.  Tried the config/reinstall a few times.

Not sure what command output I can include to diagnose, here is lsusb -v for what I think is the camera.

```
Bus 001 Device 003: ID 0c45:6d14 Microdia Integrated_Webcam_HD
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0c45 Microdia
  idProduct          0x6d14 
  bcdDevice           92.67
  iManufacturer           2 Sonix Technology Co., Ltd.
  iProduct                1 Integrated_Webcam_HD
```

andrewmv, 2020-06-18:
I am successfully using Howdy on a 2020 Dell XPS 13 9300, though I'm using Ubuntu 18.04. I didn't need to change any of the default configuration in order for the camera to be recognized, though I did need to tweak the darkness settings for it to get successfully detect a face - my open PR #366 deals specifically with this issue.

Your problem description makes me suspect your RGB camera might have been detected as infrared. Try explicitly defining the camera device - you can use the following command with each of the devices in /dev/video* to try to determine which video device is correct:
`fswebcam -d /dev/videoX out.jpg`

boltgolt, 2020-06-21:
Agreed @andrewmv , have you tried this @ctrlbrk42 ?

FIUSHerson, 2020-06-27:
I actually have the same device running Pop, and the tweak I made to get this working was to enter `sudo howdy config`, and changing `device_path = /dev/video2`.

In case that specific device path doesn't work, you can try searching up your list of webcams and ir sensors with `ls -ltrh /dev/video*`

Hope this helps!

GarvTambi, 2020-07-14:
Sometimes face detect successfully and sometimes it shows Unknown Error:1 Why?
Please help!

[image]

DasJott, 2020-07-15:
I also run a 2020 xps13 9300 on Manjaro Linux and it doesn't work out of the box.
In config the path for webcam is `none`.
How do I get the correct path for the IR cam?

andrewmv, 2020-07-19:
You'll need to determine the appropriate device path for your webcam under your distribution - @FIUSHerson 's advice above should get you going.

DasJott, 2020-07-20:
I'm sorry, I could have seen that solution.
Put `/dev/video2` into config and it works.
Thank you anyway for your answer!

ppraisethesun, 2020-08-01:
Same problem here with this device. IR emitters do not turn on, so the image is too dark for howdy to recognize a face

blackbicbiro, 2020-08-03:
Likewise. IR does not enable when using video2. I have resulted in using video0 which i think is just the normal camera, Not IR. Not Ideal. I have used my phone camera to double check that the IR is not coming on.

boltgolt, 2020-08-10:
@ppraisethesun @blackbicbiro  You both could have unsupported cameras if that's the case. What does `sudo howdy test` show when you use the IR camera? Can you view IR camera output in other applications such as cheese?

blackbicbiro, 2020-08-24:
howdie test works as a black/white camera but the IR doesnt seem to illuminate. I also used a phone camera to check to see if they IRs light up. The camera is self does have IR.
If i try to open the camera in cheese i get Horizontal striping overlayed over the picture
If i load the camera in VLC i get a very very dark picture. pretty much only see Lightbulbs
Rob

Danhyal-Masood, 2020-08-26:
what model do you have? seems dell might be using different camera depending on the model or region. Output from lsusb -v on the 9500 1650 ti model.
```
Bus 001 Device 004: ID 0bda:5510 Realtek Semiconductor Corp. Integrated_Webcam_HD
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x5510 
  bcdDevice           91.79
  iManufacturer           3 CN02FP438LG0005UA3LMA01
  iProduct                1 Integrated_Webcam_HD
```

blackbicbiro, 2020-08-27:
output from lsusb -v
XPS 15 9500 GTX 1650 Ti 
clearly right with different models camera depending on where they are being shipped
 
```
Bus 001 Device 003: ID 0c45:6d14 Microdia Integrated_Webcam_HD
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0c45 Microdia
  idProduct          0x6d14 
  bcdDevice           92.67
  iManufacturer           2 
  iProduct                1 
  iSerial                 0 
  bNumConfigurations      1
```

KalasLavas, 2020-09-18:
I also have issue with IR emitters, while IR camera works perfectly.

[image]
_White dot is tv remote_

`lsusb -v`
`Bus 001 Device 004: ID 0c45:6d14 Microdia Integrated_Webcam_HD
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0c45 Microdia
  idProduct          0x6d14 
  bcdDevice           92.67
  iManufacturer           2 
  iProduct                1 
  iSerial                 0 
  bNumConfigurations      1`

abraha2d, 2020-12-18:
I'm having the same issue on an XPS 13 9310 developer edition. Howdy works with the regular camera at /dev/video0, but /dev/video2 just results in a really dark image. The IR emitters are not turning on.

One thing I noticed is that there is only 1 physical camera, with multiple streams.

`Bus 003 Device 003: ID 0c45:672a Microdia Integrated_Webcam_HD`

```
$ v4l2-ctl --list-devices
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-6):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```

Of these, only /dev/video0 and /dev/video2 work with `fswebcam`.

EmixamPP, 2020-12-18:
I think my repo :  https://github.com/EmixamPP/linux-enable-ir-emitter can help you to solve your problems, a detailed tutorial can be found in the readme, as well as the scripts necessary to activate your infrared emitter.

abraha2d, 2020-12-19:
@EmixamPP That did it! Thank you.

For others, these are the values that work on my XPS 13 9310 DE:
```
    __u8 data[9] = {0x01,0x03,0x02,0x00,0x00,0x00,0x00,0x00,0x00};
```
```
    struct uvc_xu_control_query query = {
        .unit = 0x04, //2 first symbols of wIndex
        .selector = 0x06, //2 first symbols of wValue
        .query = UVC_SET_CUR,
        .size = 9, //wLength
        .data = (__u8*)&data,
    };
```

KalasLavas, 2020-12-19:
@abraha2d Thanks, same `.unit` value worked for XPS 15 9500

abraha2d, 2021-01-08:
Update: I got a replacement XPS 13 9310 DE, and it works out of the box with howdy. It has a `Realtek Semiconductor Corp. Integrated_Webcam_HD` instead of a `Microdia Integrated_Webcam_HD`.

kamushadenes, 2021-01-19:
For my XPS 13 9300, I had to set `dark_threshold = 80` to get any reliable results. Curiously the IR sensors didn't activate otherwise.

appoli13, 2021-01-26:
> Update: I got a replacement XPS 13 9310 DE (original one had a weird French/English keyboard), and it works out of the box with howdy. It has a `Realtek Semiconductor Corp. Integrated_Webcam_HD` instead of a `Microdia Integrated_Webcam_HD`.

This is very interesting - I have one of the original XPS 9310 DEs (ordered around the day it was released & was delivered a few weeks later). I just double checked and I have the Microdia integrated webcam (I honestly had thought that the device was by Realtek - probably got this through Dell's pdf specs...) & I have an international keyboard as well; the one with the layer 2/3 sub-keys and such, but thought that was a caveat of the whole "developer addition" thing.
I actually have had the motherboard on my machine replaced as the device was doing full hardware freezes whilst under battery power. As I had the pro plus warranty someone just came to my home and swapped it out for me, but now I'm kind of wondering if a full swap would have been better overall (I know the original stock didn't have the option for 32 gb of ram so I assume that might be one of the other changes?)

If I may ask what was the reason for you having your machine replaced? And are there any other changes that you were able to discern? 
I'll give the IR enable script and see how it works out in various use cases, but it's not a huge hassle thankfully. Depending on what I find/further info maybe I'll press Dell a bit to get my unit replaced (especially if their specs state specifically that the camera is a realtek device).

I must say I was getting quite frustrated back in sept/oct with the IR emitters & the previous script that was mentioned not working for me! Glad to see that more succinct solutions have come forth :) thank you all!

Thanks!

abraha2d, 2021-01-27:
> This is very interesting - I have one of the original XPS 9310 DEs (ordered around the day it was released & was delivered a few weeks later). I just double checked and I have the Microdia [...]

The keyboard was the only thing wrong with it. A bit confusing, and it got really annoying after a few days, so I figured I'd return/buy a new one before it was too late (since they told me their system wasn't letting them process a replacement with a different keyboard).

The only other thing that was different (that I noticed) was the SSD. The old one had a Dell-branded SSD inside (Toshiba under the label), while this one has a Kioxia SSD. Basically the same company so I thought it was just a naming thing, but when they sent out a replacement SSD for the original laptop, it was the same Dell-branded Toshiba SSD. ¯\_(ツ)_/¯

Not sure what the Toshiba SSD showed up as, but this Kioxia one shows up as `01:00.0 Non-Volatile memory controller: KIOXIA Corporation Device 0001` in lspci.

creimer808, 2023-01-06:
@EmixamPP that worked so well. thank for an amazing app :)