boltgolt/howdy #371: Howdy 2.5.1 prohibits logging in to Kubuntu 20.04 after startup

Juho-of-Consumerium-org, 2020-06-23:
Hello.

I had a rather unpleasant experience when I enabled Howdy: 

**Problem**: After setting it up and logging into the Kubuntu after startup I enter my password, the **red infrared lights fire up once**, Howdy fails to recognize me and the password field is grayed out and nothing can be done, except power off machine.

My machine is a Lenovo T480s. 

First I ran `sudo apt install v4l-utils` and ran it to get info about my cameras to try to figure out what is wrong

I needed to change the following lines in the conf to get it to work (when logged in it lets me run sudo commands with only my face):

```
    max_height = 400
    frame_width = 400
    frame_height = 400
```
Howdy started to work for 'sudo'-authentication.

So it would **seem** to me that when I enter the password to login, this authorizes the Howdy to try the face recognition one time, but then it stops (or more likely is stopped from) trying to identify the face.

Luckily for me I had Timeshift installed and active, so I was able to revert to a backup by installing and running Timeshift from a live USB stick.

I just upgraded Howdy to 2.6.0.

If I forgot some relevant info from here, please request any extra info from me.

Cheers, Juho

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Kubuntu 20.04

Howdy version: 2.5.1

ralphocdol, 2020-06-23:
I had the exact same issue recently on my kubuntu 20.04.

Go to howdy config and set:
`no_confirmation = true`

A tip on your `rebooting the machine I was rendered unable to login to Kubuntu`, just go to `Alt+F1` or the CLI session of your kubuntu and run `sudo howdy disable 1` to disable the howdy (facial recognition  works here) and login.

boltgolt, 2020-06-23:
Thanks for your very detailed report! Some lockscreens have a rare bug where they will crash if a PAM module prints anything to console, using the fix suggested by @ralphocdol would work if that's the case. It might also be a conflict an encrypted home directory, is your home encrypted? 

These freezes normally only happen in the graphical interface, pressing Ctrl + Alt + F1 should drop you to a text TTY if you're stuck

Juho-of-Consumerium-org, 2020-06-24:
Thank you @ralphocdol and @boltgolt for your prompt responses. 

Setting `no_confirmation = true` fixes the login to Kubuntu GNU/Linux OS problem.

Would there be a way to set the `no_confirmation` variable so that confirmation is showed when already logged into the system and firing sudo, but that it would not be on when logging into the OS?

Yet I am a little worried as I cannot seem to be able to access CLI with either` ALT+F1` or `Ctrl + Alt + F1`. Perhaps this is due to the Lenovo T480s requiring the `Fn-key` to be pressed to access the function keys, without the Fn-key the F-keys are multimedia control buttons. I've been trying to search for this problem, but haven't seen an exact match

Also when from the Grub2-menu I choose "rescue mode" and then "root shell" it does not let me run `sudo howdy disable 1`, so I'm kind of freaked out about the risk of locking myself out of the system again. 

Is there some way to disable howdy if it breaks and prohibits me from logging in? A way that would not be accessible easily to an intruder (in case my laptop gets stolen)

Using Howdy 2.6.0 now on Kubuntu 20.04

boltgolt, 2020-06-25:
What might be nicer for you is to disable Howdy for login, but enable it for sudo or other places you want to use it. See [this wiki page](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types) if that would work for you

amcolash, 2020-08-02:
Ubutuntu variants tend to use `F1` for the primary TTY. Use `Ctrl + Alt + F2`

einsteinarbert, 2023-08-19:
I detected that ubuntu 22.04 (kde) using SDDM service for login. When it stuck on login screen after `enter` with/without any password, I must to restart sddm: `systemctl restart sddm.service`
So I google search answer with `sddm` and found the good: [closed issue](https://github.com/boltgolt/howdy/issues/219)
=> I edited my `/etc/pam.d/sddm`, added line below:
> auth sufficient pam_python.so /lib/security/howdy/pam.py
```config
cat /etc/pam.d/sddm
#%PAM-1.0

# Block login if they are globally disabled
auth    requisite       pam_nologin.so
auth    required        pam_succeed_if.so user != root quiet_success

# auth    sufficient      pam_succeed_if.so user ingroup nopasswdlogin
auth sufficient pam_python.so /lib/security/howdy/pam.py
@include common-auth
.......
```
When I restart my laptop, the camera is turned on after press `enter` and sddm allowed me to login without any Stuck like before :100: 
So I think anyone who has been stuck like this, can fix in /etc/pam.d/ like me.
I also run this command for sure it work:
> sudo chmod -R 755 /lib/security/howdy

closed 2026-11-16 by boltgolt