boltgolt/howdy #372: On facial recognition fail, ask for password

ralphocdol, 2020-06-23:
This was on version 2.5.1, though I believe it was not a feature but rather a behavior.

It would be nice to have it as an option when if it fails or the user deliberately prevents, stops or skip the facial recognition then ask for a sudo password instead rather than waiting for the timeout or the number of times it will ask.

This is good specially at night where your cheap camera can't see your face.

Hoping to see this on next updates.

I'm on version 2.6.0

boltgolt, 2020-06-23:
It still does exactly what you're describing unless you have a non-default PAM setup

ralphocdol, 2020-06-23:
I believe I did not touch anything except `no_confirmation` in the config. It was working before I update to 2.6.0

GarvTambi, 2020-07-14:
Sometimes face detect successfully and sometimes it shows Unknown Error:1 Why?
Please help!

[image]

closed 2021-04-02 by boltgolt