boltgolt/howdy #373: Lenovo L13 Nonworking IR Blasters

ccwky11, 2020-06-23:
_Please describe the issue in as much detail as possible, including any errors and traces_

Attempting to set up the IR camera and it does not appear to work, the IR blasters do not initiate in any of the cameras (there are 5 /dev/video# options, of which only 2 do not throw an error on start, video0 and video2). any help would be greatly appreciated.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):
Manjaro 20
Howdy version:
2.5.1-2

boltgolt, 2020-06-23:
Unfortunately that probably means your IR camera is unsupported, can you open that camera in other applications?

ccwky11, 2020-06-23:
The camera itself activates, however it appears that the IR blasters do not, here is an example image:
[image]

ccwky11, 2020-06-23:
when running the test command found here: "gst-launch-1.0 v4l2src device=/dev/video2 num-buffers=10 ! image/jpeg ! multifilesink location="frame-%02d.jpg"

I get the following output:

Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../gstreamer/libs/gst/base/gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
ccwky12@Manjaro-L13 ~ [1]>

ccwky11, 2020-06-23:
Found a fix for the issue here:
https://github.com/boltgolt/howdy/issues/312

while not perfect, it does provide a band-aid solution, thank you so much for your help.

closed 2020-06-23 by ccwky11