boltgolt/howdy #374: Stop using broadcast / journald to return error messages

jmlgo, 2020-06-24:
_Please describe the issue in as much detail as possible, including any errors and traces_

Since the last update, every time howdy is attemping an authentication, I get notifications (sound, system popup) and ALL my terminals get messed with a broadcast message, even when `detection_notice = false` is configured.

```
Broadcast message from systemd-journald@xxxxxx (Wed 2020-06-24 12:07:18 CEST):

sudo[11610]: [HOWDY] Attempting facial authentication for user xxxxxx

Broadcast message from systemd-journald@xxxxxx (Wed 2020-06-24 12:07:20 CEST):

sudo[11610]: [HOWDY] Login approved
```

Pretty similar to this: https://github.com/s3fs-fuse/s3fs-fuse/issues/301

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Kubuntu 20.04

Kernel: 5.4.0-37-generic

Howdy version: Howdy 2.6.0

principis, 2020-06-24:
The syslog api is used incorrectly.

For example https://github.com/boltgolt/howdy/blob/master/src/pam.py#L38
```python
syslog.syslog(syslog.LOG_AUTH, "[HOWDY] Attempting facial authentication for user " + pamh.get_user())
```

The first parameter is the priority, but the `LOG_AUTH` constant is passed, which is used for the facility causing the message to be logged with emerg severity.
It should be changed to something like this, with appropriate `priority` for each message.

```python
syslog.openlog(facility=syslog.LOG_AUTH)
syslog.syslog(syslog.LOG_INFO, "[HOWDY] Attempting facial authentication for user " + pamh.get_user())
...
syslog.syslog(syslog.LOG_NOTICE, "[HOWDY] Failure, timeout reached")
...
syslog.closelog()
```

Bubsbilby, 2020-06-24:
I'm having exactly the same issue. It's also preventing me from unlocking the lock-screen with howdy, but the password still works. The login screen still works as expected however. I've used both fixes found in the common issues page.

ramiel, 2020-06-25:
I can confirm the same error

boltgolt, 2020-06-25:
It seems that using `LOG_AUTH` also sets the priority to `LOG_EMERG`, even though that's not [documented](https://docs.python.org/3/library/syslog.html) and does not happen on my system. However, using `LOG_NOTICE` causes the log messages to end up in the general system log, not the authentication log

Would using `syslog.LOG_NOTICE | syslog.LOG_AUTH` in `/lib/security/howdy/pam.py` work for everyone here?

nmaggioni, 2020-07-01:
@boltgolt That workaround works fine on Manjaro.

closed 2020-09-02 by boltgolt

boltgolt, 2020-09-02:
Finally fixed in 2.6.1, sorry it took so long. Should slowly make its way to the other distros

PR #380 (2020-08-01): Use syslog.LOG_INFO and LOG_NOTICE instead of LOG_EMERG
Fixes #374
commit 9de3d39e (2020-08-01): Use syslog.LOG_INFO and LOG_NOTICE instead of LOG_EMERG

Fixes #374