boltgolt/howdy #375: Error when adding a face (since Ubuntu 20.04)

Liemaeu, 2020-06-25:
Since I upgraded to Ubuntu 20.04, Howdy does no longer work.
I reinstalled Howdy (this worked), but when I try to add a face I get this error (after entering the name):
`Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 99, in <module>
    video_capture = VideoCapture(config)
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 45, in __init__
    self._create_reader()
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 115, in _create_reader
    self.internal = cv2.VideoCapture(
AttributeError: module 'cv2' has no attribute 'VideoCapture'
Exception ignored in: <function VideoCapture.__del__ at 0x7f05e9543700>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'NoneType' object has no attribute 'release'`

I double checked that I selected the right camera (the IR one, not the normal webcam) in during Howdy installation.

FIUSHerson, 2020-06-27:
Same issue here. In my config file, my device path was automatically set to /dev/v4l/by-path/none, indicating that something went wrong when configuring the application.

JPYamamoto, 2020-06-30:
Same here. I can't make a pull request, but let me know if there's more info I could provide in order to help.

ahsansalim200, 2020-07-01:
Same here...On Pop Os 20

FIUSHerson, 2020-07-02:
You can fix this issue by manually looking up your IR camera path, and putting it in the config file.

Seltonu, 2020-07-05:
Same issue on fresh install of Pop!_OS 20.04 (ubuntu)

[image]

mdnahas, 2020-07-08:
For newbies:

I checked which device was the IR camera using ffmpeg.  Mine was /dev/video2.
   `ffmpeg -f v4l2 -list_formats all -i /dev/video2`
The command spits a lot of output, but the last line says "Greyscale":
   `[video4linux2,v4l2 @ 0x562d4bfa59c0] Raw       :        gray :      8-bit Greyscale : 340x340`

Next, I ran
   `sudo howdy config`
and changed the value of "device_path" to "/dev/video2".

After that, "`sudo howdy add`" ran correctly.  I tested with "`sudo howdy test`".  (Actually, on my system, I need "`sudo QT_X11_NO_MITSHM=1 howdy test`".)   

If you care, I've got a Dell XPS 13 9370.

techgaun, 2020-07-08:
Ran into the same issue myself as well. I didn't even notice `howdy config` command so I went straight to `/lib/security/howdy/config.ini` and modified the `device_path` to point to the right one (in my case, video0).
Alternatively, you can use `v4l2-ctl --list-devices` to get the camera devices (install via `sudo apt install v4l-utils`).

techgaun, 2020-07-09:
FWIW, its probably a good idea to handle the exception and let the user know better information (optionally, including quick how-tos). I could do a quick PR if we can agree on this.

yohaoquan, 2020-07-10:
I second that of what @techgaun said. Plus, it's also clear that the program should detect and select a valid path pointing to the IR device and now it's not on Ubuntu 20.04. 
So not only adding catch, but also improve the camera detection at install time or init is needed.

Liemaeu, 2020-07-11:
@mdnahas I tried this (I did "ls /dev/* | grep video" and "ffmpeg -f v4l2 -list_formats all -i" for all /dev/video [/dev/video0 - /dev/video3 in my case] and only /dev/video2  outputted the "gray". So I ran "sudo howdy config", and replaced the "device_path"-line with "device_path = /dev/video2), but I still get this error when I try to add a face:
`Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 99, in <module>
    video_capture = VideoCapture(config)
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 45, in __init__
    self._create_reader()
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 115, in _create_reader
    self.internal = cv2.VideoCapture(
AttributeError: module 'cv2' has no attribute 'VideoCapture'
Exception ignored in: <function VideoCapture.__del__ at 0x7f3522d28700>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'NoneType' object has no attribute 'release'`

GarvTambi, 2020-07-14:
Sometimes face detect successfully and sometimes it shows Unknown Error:1 Why?
Please help!

[image]

boltgolt, 2020-07-14:
Final warning @GarvTambi, stop spamming

boltgolt, 2020-08-09:
An exception handler could be added there, maybe linking to a wiki page?

FIUSHerson, 2020-08-09:
I suggest having the camera setup the way it was before (asked if it was the correct camera) but done when adding a face but only for the first time. I also suggest adding an argument for that as well.

wesleyhixon, 2020-08-09:
Everything is set up correctly (i think) and my razer blade 15 running mint 20 won't add a face because it failed to read camera specified in device_path.
I set my device path to dev/video2 and the ir camera turns on but it still gives me the same error

nalin29, 2020-08-09:
You need to change the image size my razor blade was 360 by 360 I posted my settings on the got wiki take a look.

wesleyhixon, 2020-08-09:
It worked!!!!!! Thanks!

> You need to change the image size my razor blade was 360 by 360 I posted my settings on the got wiki take a look.

amoscao, 2020-12-29:
This worked for me too. Specifically, I'm using a Logitech Brio which had 4 devices: /dev/video0, /dev/video1, /dev/video2, and /dev/video3. video0 corresponded to the regular webcam, while video2 was the IR.

zqpie, 2021-01-05:
> It worked!!!!!! Thanks!
> 
> > You need to change the image size my razor blade was 360 by 360 I posted my settings on the got wiki take a look.
does anyone know how do i do that?

zqpie, 2021-01-05:
> It worked!!!!!! Thanks!
> does anyone know how i can do that?

> > You need to change the image size my razor blade was 360 by 360 I posted my settings on the got wiki take a look.

zqpie, 2021-01-05:
does anyone know how I can do that?

> It worked!!!!!! Thanks!
> 
> > You need to change the image size my razor blade was 360 by 360 I posted my settings on the got wiki take a look.

Seltonu, 2021-01-05:
I'm thinking you'll probably want to adjust frame_width and frame_height in Howdy to the resolution of your webcam, run "sudo howdy config" to edit the settings.

As for what resolution to set, you can follow the info [here](https://askubuntu.com/questions/214977/how-can-i-find-out-the-supported-webcam-resolutions) but in short, run "lsusb" to find your webcam Bus and Device id then put them into this command **lsusb -s [your device bus here]:[your device ID here] -v | egrep "Width|Height"**. The output is in pairs

Use smaller resolutions in your howdy config for more performance at the cost of accuracy, I'd assume 480p or 720p would be just fine(?)

zqpie, 2021-01-05:
I got it figured out, for anyone with the same issue, make sure the “video2” in howdy config is replaced with the correct one for you pc, mine was video0
> I'm thinking you'll probably want to adjust frame_width and frame_height in Howdy to the resolution of your webcam, run "sudo howdy config" to edit the settings.
>
> As for what resolution to set, you [...]

I got fig

juandisay, 2021-02-08:
you can try change `device_path = /dev/video0`
it works for me in ubuntu 18.04 LTS

closed 2021-04-02 by boltgolt

ALENTL, 2021-04-25:
> Same here...On Pop Os 20

same here dude

FredyVia, 2021-09-24:
In my case(Lenovo Xiaoxin pro13  with Ubuntu 20.04), after the above steps (video2 width height), sudo pip install opencv-contrib-python is also carried out before it runs correctly. hope that helps later people.

ChrisTech77, 2024-06-21:
Error when adding a face Ubuntu 24.04

my face unlock perfectly fine with win 10, got error on ubuntu

I ran 
sudo howdy add
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 96, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 11, in <module>
    from recorders.video_capture import VideoCapture
  File "/lib/security/howdy/recorders/video_capture.py", line 6, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'

Anyone knows how to fix it ?

fadelfffar, 2024-12-23:
asking the same as ChrisTech77