boltgolt/howdy #376: Adding Face Model does not work in Fedora 32

manojkiraneda, 2020-06-25:
I was following the instructions in the below link :

https://copr.fedorainfracloud.org/coprs/principis/howdy/

Every thing works fine, except Adding the Model.

`howdy test is properly invoking the IR cameras, and i was able to see the circle on the face`

But when i try to add a Face Model, i was getting a floating point exception :
```
Enter a label for this new model [Initial model] (max 24 characters): Manoj
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/videoio_registry.cpp (187) VideoBackendRegistry VIDEOIO: Enabled backends(7, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for FFMPEG
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (690) open OpenCV | GStreamer: /dev/video0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (723) open OpenCV | GStreamer: mode - FILE
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (INTEL_MFX): glob is 'libopencv_videoio_intel_mfx*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for INTEL_MFX

Please look straight into the camera
Floating point exception
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora32

Howdy version:2.5.1-4.fc32

glmdev, 2020-06-26:
I'm also experiencing this issue. I tried re-installing howdy, and I now get a dependency error when installing from the prinicpis COPR repo:

``` 
Last metadata expiration check: 0:07:34 ago on Fri 26 Jun 2020 09:32:22 AM CDT.
Error: 
 Problem: conflicting requests
  - nothing provides python3dist(dlib) >= 6.0 needed by howdy-2.5.1-4.fc32.x86_64
(try to add '--skip-broken' to skip uninstallable packages)
```

manojkiraneda, 2020-06-28:
I did try re-installing the howdy package..and hit the same issue which @glmdev has mentioned.

manojkiraneda, 2020-07-02:
@luyatshimbalanga @principis , any help on this would be really helpful.Thanks.

luyatshimbalanga, 2020-07-04:
@manojkiraneda Could you try the update and see if that worked?

principis, 2020-07-04:
@manojkiraneda Sorry, I messed up when updating dlib. Please remove howdy and python3-dlib and reinstall them. It should work fine then.

shacky, 2020-07-04:
It works now! Thanks!

manojkiraneda, 2020-07-04:
@principis yes, this works fine for sudo, but when i use howdy to unlock the gdm - screen lock, it is throwing an Unkown error:-8, Any idea of what this is about ?

Also i see that, the repo - still uses 2.5.1 , are you planning to migrate to 2.6 ?

Following are the relevant journal traces:

```
Jul 04 19:09:58 juliet gdm-session-worker[13985]: [ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/videoio_registry.cpp (187) VideoBackendRegistry VIDEOIO: Enabled backends(7, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940)
Jul 04 19:09:58 juliet gdm-session-worker[13985]: [ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
Jul 04 19:09:58 juliet gdm-session-worker[13985]: [ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
Jul 04 19:09:58 juliet gdm-session-worker[13985]: [ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for FFMPEG
Jul 04 19:09:58 juliet gdm-session-worker[13985]: [ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (690) open OpenCV | GStreamer: /dev/video0
Jul 04 19:09:58 juliet gdm-session-worker[13985]: [ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (723) open OpenCV | GStreamer: mode - FILE
Jul 04 19:09:58 juliet gdm-session-worker[13985]: [ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
Jul 04 19:09:58 juliet gdm-session-worker[13985]: [ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
Jul 04 19:09:58 juliet gdm-session-worker[13985]: [ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Jul 04 19:09:58 juliet audit[13985]: ANOM_ABEND auid=1000 uid=0 gid=1000 ses=1 pid=13985 comm="python3" exe="/usr/bin/python3.8" sig=8 res=1
Jul 04 19:09:58 juliet kernel: traps: python3[13985] trap divide error ip:7f690c85988f sp:7ffead27ede0 error:0 in libopenblaso-r0.3.9.so[7f690c76f000+20d0000]
Jul 04 19:09:58 juliet kernel: audit: type=1701 audit(1593869998.886:1361): auid=1000 uid=0 gid=1000 ses=1 pid=13985 comm="python3" exe="/usr/bin/python3.8" sig=8 res=1
Jul 04 19:09:58 juliet audit: BPF prog-id=304 op=LOAD
Jul 04 19:09:58 juliet kernel: audit: type=1334 audit(1593869998.896:1362): prog-id=304 op=LOAD
Jul 04 19:09:58 juliet kernel: audit: type=1334 audit(1593869998.896:1363): prog-id=305 op=LOAD
Jul 04 19:09:58 juliet audit: BPF prog-id=305 op=LOAD
Jul 04 19:09:58 juliet systemd[1]: Started Process Core Dump (PID 14010/UID 0).
Jul 04 19:09:58 juliet audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@18-14010-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jul 04 19:09:58 juliet kernel: audit: type=1130 audit(1593869998.897:1364): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@18-14010-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jul 04 19:09:59 juliet systemd-coredump[14011]: Process 13985 (python3) of user 0 dumped core.
                                                
                                                Stack trace of thread 13985:
                                                #0  0x00007f690c85988f sgemm_thread_tn (libopenblaso.so.0 + 0x21a88f)
                                                #1  0x00007f690928d136 cblas_sgemm (libopenblasp.so.0 + 0x139136)
                                                #2  0x00007f6907d339db n/a (/home/juliet/.local/lib/python3.8/site-packages/dlib.cpython-38-x86_64-linux-gnu.so + 0x6759db)
                                                #3  0x00007f6907d3498d n/a (/home/juliet/.local/lib/python3.8/site-packages/dlib.cpython-38-x86_64-linux-gnu.so + 0x67698d)
                                                #4  0x00007ffead27f148 n/a (n/a + 0x0)
                                                #5  0x0000000000000020 n/a (n/a + 0x0)
Jul 04 19:09:59 juliet systemd[1]: systemd-coredump@18-14010-0.service: Succeeded.
Jul 04 19:09:59 juliet audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@18-14010-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jul 04 19:09:59 juliet kernel: audit: type=1131 audit(1593869999.513:1365): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@18-14010-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jul 04 19:09:59 juliet audit: BPF prog-id=305 op=UNLOAD
Jul 04 19:09:59 juliet audit: BPF prog-id=304 op=UNLOAD
Jul 04 19:09:59 juliet kernel: audit: type=1334 audit(1593869999.591:1366): prog-id=305 op=UNLOAD
Jul 04 19:09:59 juliet kernel: audit: type=1334 audit(1593869999.591:1367): prog-id=304 op=UNLOAD
Jul 04 19:10:00 juliet abrt-server[14017]: Package 'howdy' isn't signed with proper key
Jul 04 19:10:00 juliet abrt-server[14017]: 'post-create' on '/var/spool/abrt/ccpp-2020-07-04-19:09:59.808835-13985' exited with 1
Jul 04 19:10:00 juliet abrt-server[14017]: Deleting problem directory '/var/spool/abrt/ccpp-2020-07-04-19:09:59.808835-13985'
Jul 04 19:10:04 juliet gdm-password][13981]: gkr-pam: the password for the login keyring was invalid.
```

principis, 2020-07-04:
@manojkiraneda I think you are using the wrong dlib version, it should be 19.19 (19.20 is broken and I have no idea why). Can you check with pip3 list or a different way?

I decided not to upgrade to 2.6.0 because of #374

manojkiraneda, 2020-07-04:
@principis Thanks alot , yes infact i was using dlib - 19.20 which is causing the issue, i have removed it , and now it all looks good.

manojkiraneda, 2020-07-04:
> @manojkiraneda I think you are using the wrong dlib version, it should be 19.19 (19.20 is broken and I have no idea why). Can you check with pip3 list or a different way?
> 
> I decided not to upgrade to 2.6.0 because of #374

closed 2020-07-04 by manojkiraneda

principis, 2020-07-08:
@manojkiraneda I've created a patched version of 2.6.0, you can find the git repo over [here](https://github.com/principis/howdy/tree/patched) and the copr repo over [here](https://copr.fedorainfracloud.org/coprs/principis/howdy-patched/)