boltgolt/howdy #377: Howdy to unlock SSH key?

ryanerwin, 2020-06-30:
Any thoughts on how to connect Howdy an ssh-agent or otherwise use Howdy to unlock SSH keys?

Would be preferable to sit in front of my monitor to decrypt my SSH keys rather than repeatedly typing in a password...

A bit more fun than keeping everything in the keychain.

How difficult to connect Howdy to other apps or embed Howdy in other apps?

GarvTambi, 2020-07-14:
Sometimes face detect successfully and sometimes it shows Unknown Error:1 Why?
Please help!

[image]

boltgolt, 2021-04-02:
I believe your SSH keys are encrypted with your password, so you will always have to enter your password to decrypt them. There's no way Howdy can be of use here unfortunately

closed 2021-04-02 by boltgolt

nixigaj, 2025-02-21:
If you use disk encryption and use a SSH key password as a way to protect it from rouge applications running as your user or some person having physical access to your device when logged in, you can store your `.ssh` folder in another users home directory, that you then use sudo to get to. I created a user `sshuser` and moved my unencrypted `.ssh` folder there, and then i have the alias `alias ssh-example="sudo -u sshuser ssh user@example.com"`. Then you can use Howdy auth through sudo.