boltgolt/howdy #378: Update 2.6 - xscreensaver stopped working (+solution)

ju916, 2020-06-30:
**Problem**
Since the (automatic) upgrade to 2.6, unlocking xscreensaver with howdy fails:

```
xscreensaver: 13:44:44:   pam_authenticate (...) ...
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 23, in <module>
    from recorders.video_capture import VideoCapture
ModuleNotFoundError: No module named 'recorders.video_capture'

```

The weird thing is, that sudo for example works with the same configuration.
That brings me to the core of the problem: unlike sudo xscreensaver runs with my uid -- and does not have access to the directory recorders/:

```
root@ju-PC:/lib/security/howdy# ls -al recorders/
insgesamt 80
drwxr--r-- 3 root root  4096 Jun 30 13:36 .
```
Knowing this, the solution is straight forward:

**FIX:**

`# chmod a+x recorders
`

works for me ...

Digisaurus, 2020-07-13:
Thanks for your issue report [ju916](https://github.com/ju916)!

I had the same problem of xscreensaver lock generating the error with howdy after the upgrade:

`Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 23, in <module>
    from recorders.video_capture import VideoCapture
ModuleNotFoundError: No module named 'recorders.video_capture'`

The same chmod solved the problem for me:
`$ sudo chmod a+x /lib/security/howdy/recorders`

The thing that confuses me is although it's been some time since I installed xscreensaver (after having first installed Howdy), I don't recall ever having to modify permissions to get them to work together...?

crowlogic, 2023-03-21:
Confirmed,  sudo chmod a+x /lib/security/howdy/recorders did the trick. someone should fix the debian build scripts it seems easy enough

crowlogic, 2024-03-11:
Idgaf

closed 2024-04-01 by musikid