boltgolt/howdy #385: Version 2.6.0 "Broadcast message from systemd-journald" problem on Manjaro

bobzhu418, 2020-07-04:
_Please describe the issue in as much detail as possible, including any errors and traces_

After upgrading to version 2.6.0, every time if I use sudo on terminal, I get following message. It's not fatal but very irritating. 

Broadcast message from systemd-journald@manjaro (Sat 2020-07-04 10:08:23 +08):

sudo[2580]: [HOWDY] Attempting facial authentication for user bob

Broadcast message from systemd-journald@manjaro (Sat 2020-07-04 10:08:25 +08):

sudo[2580]: [HOWDY] Login approved

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Manjaro 20.0.3 Kernel 5.7.0-3

Howdy version: 2.6.0

bobzhu418, 2020-07-04:
Ok, I found a work-around. Set `ForwardToWall=no` in `/etc/systemd/journald.conf` to disable the broadcasting.

GarvTambi, 2020-07-14:
Sometimes face detect successfully and sometimes it shows Unknown Error:1 Why?
Please help!

[image]

boltgolt, 2020-08-07:
Thanks for reporting! To keep discussion about it in one place i'm going to close this now, see #374

closed 2020-08-07 by boltgolt