boltgolt/howdy #387: Privilege detection is broken when using PolicyKit

aleixpol, 2020-07-06:
I've been looking into implementing a Plasma front-end for Howdy (https://invent.kde.org/apol/plasma-howdy) although it's not working. For escalating privileges like we need for howdy (which is something we probably should be revised sometime) we generally use PolicyKit, but when we do so howdy complains about not being run under root for some reason.

An easy way to test it is this:
```
$ pkexec python /lib/security/howdy/cli.py list
Please run this command as root:

        sudo howdy list
```
----

Linux distribution (if applicable): ArchLinux
Howdy version: 2.6.0

GarvTambi, 2020-07-14:
Sometimes face detect successfully and sometimes it shows Unknown Error:1 Why?
Please help!

[image]

aleixpol, 2020-07-14:
Please open a separate issue for this, it's unrelated.

boltgolt, 2020-07-14:
Please stop spamming this in every issue, or i will have to remove you @GarvTambi

boltgolt, 2020-08-07:
A frontend for Howdy would be amazing! 

The issue is the way Howdy checks if it has been run with sufficient permissions. Right now it's a [simple check](https://github.com/boltgolt/howdy/blob/master/src/cli.py#L83) to see if the environment variable "SUDO_USER" has been set. Something more robust might be needed there.

Pointedstick, 2020-08-13:
You could maybe check to see if the UID is 0. That works for pkexec and also catches sudo too.

```
$ id -u
1000

$ pkexec id -u
0

$ sudo id -u
0
```

boltgolt, 2021-05-07:
Probably, fix is in and the GUI is done too

closed 2021-05-07 by boltgolt