boltgolt/howdy #388: Apache pwauth and Howdy - Remote compromise risk?

ryanerwin, 2020-07-09:
In my apache config I had:

```
		AddExternalAuth pwauth /usr/sbin/pwauth
		SetExternalAuthMethod pwauth pipe

		<Location />
			SSLRequireSSL
			AuthType Basic
			AuthName "Authentication"
			AuthBasicProvider external
			AuthExternal pwauth
			require valid-user
		</Location>
```

Inside of `/var/log/auth.log` this was causing:

```
Jul  9 11:21:23 ri9 pwauth: [HOWDY] Attempting facial authentication for user ryan
Jul  9 11:21:23 ri9 pwauth: [HOWDY] Failure, unknown error1
Jul  9 11:21:23 ri9 pwauth: [HOWDY] Attempting facial authentication for user ryan
Jul  9 11:21:24 ri9 pwauth: [HOWDY] Failure, unknown error1
Jul  9 11:21:24 ri9 pwauth: [HOWDY] Attempting facial authentication for user ryan
Jul  9 11:21:25 ri9 pwauth: [HOWDY] Failure, unknown error1
Jul  9 11:21:25 ri9 pwauth: [HOWDY] Login approved
```

I can be sure these were apache caused, not only because of the `pwauth` line, but also because the moment I stopped Apache, no additional log entries were added.

If you were allowing pam based authentication to sshd, not sure if you could also get howdy to attempt to accept a remote user's login just because you're in front of the camera.

Using `vsftpd`, I also have pam authentication running, and that also seems to connect pam to howdy for authentication... Inside of `/etc/vsftpd.conf` I have:

```
pam_service_name=vsftpd
```

In the log, that still generates:
```
Jul  9 12:04:22 ri9 vsftpd: [HOWDY] Attempting facial authentication for user ryan
```

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 20.04

Howdy version: v2.6.0 (installed via apt, http://ppa.launchpad.net/boltgolt/howdy/ubuntu)

ryanerwin, 2020-07-12:
Going through some other tickets, I think I've found a way to solve this.

If I were the maintainer of `howdy`, I don't think I would ship it with howdy updating `auth-common`, as it seems to me any PAM remote authentication is automatically validated by you sitting in front of your webcam...

I'm sure this fix isn't ideal, but I don't want to lock myself out with a PAM mis-configuration... 

First, back up the Howdy version of `auth-common`
```
cp /etc/pam.d/common-auth /etc/pam.d/common-auth-howdy
```
Then I did `pam-auth-update` like in #243 
```
rm /usr/share/pam-configs/howdy
pam-auth-update --package
```

Then like suggested in #176 I only added howdy to the mechanisms where we really want to use howdy:

Then just update the `@include common-auth` to `@include common-auth-howdy` in the files where we know we're assuming a local user... In my case that's only:

```
/etc/pam.d/sudo
/etc/pam.d/gdm-password
```

This notably excludes `howdy` from authentication for things like: `samba`, `sshd`, `vsftpd`, or the `pwauth` service connected to apache.

ryanerwin, 2020-08-04:
Wow! Nearly one month and no response to this?

This seems like a major hole in the way Howdy works by default.

boltgolt, 2020-08-07:
Howdy is setup to be as easy to use as possible, as the goal of the project is to be the alternative for Windows Hello on Linux. For some more advanced users this does pose a problem.

By default, authentication with Howdy is blocked if a laptop lid is closed or if the user is in an SSH shell. However, as far as i know there's no real way to find out exactly what process is calling Howdy from within pam-python. It's a compromise between ease of use and total security, and i don't expect a lot of people have a setup like yours. 

I'd like to have a blacklist of processes to ignore on auth call, but i do not think this is possible at the moment.

You didn't really have to go through other issues, your second post is basically [this wiki page i wrote in april](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types). Howdy is not my full-time job but a hobby, it can sometime take a little longer before i answer issues on here.