boltgolt/howdy #389: [Feature request] Having a prompt pop up when facial detection is underway, just like when asking for password.

yohaoquan, 2020-07-10:
I have been using the program for a while and have been loving the time it saved for me. But sometimes I still find it a bit intrusive since the camera is positioned that I don't need to explicitly turn my head in order for detection to pass. also, the camera I am using, Logitech BRIO, does not lit up its indicator when the IR camera is active. They only way I can tell that it's in use is by the faint IR emitter lighting up.

This is convenient, but also posted some greater security concerns. I am not afraid of data leaks, but of that when I made a mistake, or a program behaving suspiciously(sudoing when it should not be), I don't have a chance to close it up. 

Sometimes, a program asks for elevation in the background and it gets automatically approved and I don't even know that happened until I bring back a terminal and see the auth requests that were made.

I understand that this may be a lot of work, but maybe spawning a window that tells the user the auth is underway, or having the user confirm that they actually want to authenticate the program after successful facial detection,  just like how Windows Hello does it.

Love the work you guys are doing!

ryanerwin, 2020-07-12:
Might be possible to get `notify-send` working when called from `PAM`... Worst case scenario, could probably develop a desktop app and communicate it over a socket.

For a quick test run, might try adding a os.system call to execute a shell command line
 ```
$DISPLAY=:0 notify-send "howdy" "attempting facial recognition"
```
once the camera connection is initialized...

hanaral, 2020-07-13:
The best option imo would be to ask DE devs to make their own UX. The elementary OS team are currently tossing around a design for biometric and nitro/yubikey UX, and I'm sure that would lead the way for other auth methods like Howdy. Maybe someone could prod a few lead devs for different DEs to get their thoughts. (I'll see what Cassidy from elementary thinks)

GarvTambi, 2020-07-13:
Sometimes face detect successfully and sometimes it shows Unknown Error:1 Why?
Please help!

[image]

yohaoquan, 2020-07-16:
@hanaral Great idea. I saw your issue in elementary OS's repo. That is definitely the ultimate way of implementing it. But with my limited experience, with a distro that's large as Ubuntu, I could see this issue/feature request being assigned a low priority easily. Since it meant that a proper "auth plugin" support on Ubuntu's part needs to be done, and only after that, howdy team can start implementing itself as a plugin. With the update cycle of howdy (we are looking at 20-ish days per bugfix/feature) and it of Ubuntu's, I can easily see this whole process taking at least a year to be implemented. So maybe a solution that requires less collaboration between different teams is better at this point(?). Purely speculating here.

boltgolt, 2020-09-28:
The prompt in the the works

boltgolt, 2026-11-16:
Hey, thanks for suggesting this! v3 now shows a small popup window while Howdy is scanning (when howdy-gtk is installed), so it's clear when the camera is being used. You can also turn on `detection_notice` in the config for a text notice, and use rubber stamps (e.g. `hotkey`) if you want to confirm each authentication. Closing this as implemented!

closed 2026-11-16 by boltgolt