boltgolt/howdy #390: can't do "sudo howdy test"

Xerdiosa, 2020-07-10:
_Please describe the issue in as much detail as possible, including any errors and traces_

every time i try to test howdy, i will got:
```
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 66, in <module>
    cv2.setMouseCallback("Howdy Test", mouse)
cv2.error: OpenCV(4.3.0) /io/opencv/modules/highgui/src/window_QT.cpp:717: error: (-27:Null pointer) NULL window handler in function 'cvSetMouseCallback'
```
i can still use howdy to login normally

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu Budgie 20.04

Howdy version: 2.6.0

Xerdiosa, 2020-07-10:
i didn't do anything and it works now

Xerdiosa, 2020-07-10:
after a few test i conclude that "sudo howdy test" works with 50% success rate
```
─akbar@AKBAR-UX433FN ~ 
╰─$ sudo howdy test     

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 66, in <module>
    cv2.setMouseCallback("Howdy Test", mouse)
cv2.error: OpenCV(4.3.0) /io/opencv/modules/highgui/src/window_QT.cpp:717: error: (-27:Null pointer) NULL window handler in function 'cvSetMouseCallback'

╭─akbar@AKBAR-UX433FN ~ 
╰─$ sudo howdy test                                                         1 ↵

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

^C
Closing window
╭─akbar@AKBAR-UX433FN ~ 
╰─$ sudo howdy test

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 66, in <module>
    cv2.setMouseCallback("Howdy Test", mouse)
cv2.error: OpenCV(4.3.0) /io/opencv/modules/highgui/src/window_QT.cpp:717: error: (-27:Null pointer) NULL window handler in function 'cvSetMouseCallback'

╭─akbar@AKBAR-UX433FN ~ 
╰─$ sudo howdy test                                                         1 ↵

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

v^C
Closing window
╭─akbar@AKBAR-UX433FN ~ 
╰─$ sudo howdy test

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 66, in <module>
    cv2.setMouseCallback("Howdy Test", mouse)
cv2.error: OpenCV(4.3.0) /io/opencv/modules/highgui/src/window_QT.cpp:717: error: (-27:Null pointer) NULL window handler in function 'cvSetMouseCallback'

╭─akbar@AKBAR-UX433FN ~ 
╰─$ sudo howdy test                                                         1 ↵

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

^C
Closing window
╭─akbar@AKBAR-UX433FN ~ 
╰─$ sudo howdy test

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 66, in <module>
    cv2.setMouseCallback("Howdy Test", mouse)
cv2.error: OpenCV(4.3.0) /io/opencv/modules/highgui/src/window_QT.cpp:717: error: (-27:Null pointer) NULL window handler in function 'cvSetMouseCallback'
```
i don't think it will be a major issue, but i will leave this open just in case

GarvTambi, 2020-07-13:
Sometimes face detect successfully and sometimes it shows Unknown Error:1 Why?
Please help!

[image]

choyrim, 2020-07-16:
I resolved this by uninstalling the pip-installed `opencv-python` (4.3.0.36) and installing the apt package `python3-opencv` (4.2.0+dfsg-5).
```bash
$ sudo pip uninstall opencv-python
Found existing installation: opencv-python 4.3.0.36
Uninstalling opencv-python-4.3.0.36:
.  .  .
  Successfully uninstalled opencv-python-4.3.0.36
$ sudo apt install python3-opencv
. . .
```
so seems like the 4.3 opencv python module is not compatible with the 4.2 opencv libraries. I found the fix on the following stackoverflow
https://stackoverflow.com/questions/62801244/null-window-handler-in-function-cvsetmousecallback

My machine originally just had the `libopencv-dev` installed from apt and the `opencv-python` installed  from pip.
```bash
$ sudo apt list libopencv-dev python3-opencv
Listing... Done
libopencv-dev/focal,now 4.2.0+dfsg-5 amd64 [installed,automatic]
python3-opencv/focal 4.2.0+dfsg-5 amd64
$ pip list | grep opencv
opencv-python           4.3.0.36
$ sudo apt list howdy
Listing... Done
howdy/focal,focal,now 2.6.0 all [installed]
```

I recommend that the howdy installation package (somehow) prefer the apt `python3-opencv` over a `pip` install. perhaps installing it before letting `pip` do its thing.

nick-stocks, 2020-07-17:
I was getting this same error 50% of the time, the other 50% it worked. The solution provided works for me too - use `apt install python3-opencv`instead of `pip install opencv-python`

I now get the below warnings instead, but `howdy test` now works every time
```
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```

jkempson, 2020-07-25:
Seeing exactly the same issue. Uninstall pip opencv and installed python3-opencv but seeing the same error as @nick-stocks.

choyrim, 2020-07-27:
@nick-stocks @jkempson yeah i didn't notice those errors. i was just happy it was working for me. howdy for authentication works for me despite the errors.

I did some googling and found a fix for the gstreamer warnings. I manually modify `recorders/video_capture.py`
```diff
@@ -113,7 +113,8 @@
                else:
                        # Start video capture on the IR camera through OpenCV
                        self.internal = cv2.VideoCapture(
-                               self.config.get("video", "device_path")
+                               self.config.get("video", "device_path"),
+                               cv2.CAP_V4L
                        )
 
                # Force MJPEG decoding if true
```
which was based on the hack from here
https://github.com/opencv/opencv/issues/15074#issuecomment-576096167

I also noticed that the perms on the `recorders` directory was 744 which is strange so i chmod-ed it to 755.

If this works, then perhaps we should submit a PR or another issue with a more specific ask.

nick-stocks, 2020-07-27:
@choyrim @jkempson 
opencv hack worked to get rid of one set of warnings, and the chmod 755 fixed my other error on sudo, above ([chmod fix also found here](https://github.com/boltgolt/howdy/issues/378#issuecomment-657331212)) However howdy now not working with a new error (below). I've tried running `sudo ./install.sh`, and also as `sudo su` but still get the same error back
```
nick@pop-os:~$ sudo howdy test
Data files have not been downloaded, please run the following commands:

	cd /lib/security/howdy/dlib-data
	sudo ./install.sh

Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 224, in <module>
    face_locations = face_detector(gsframe, 1)
TypeError: 'NoneType' object is not callable
Unknown error: 1
[sudo] password for nick: 
```

nick-stocks, 2020-07-27:
Summary of fixes
```
sudo pip uninstall opencv-python
sudo apt install python3-opencv
sudo chmod 755 -R /lib/security/howdy
```
and [this hack above](https://github.com/boltgolt/howdy/issues/390#issuecomment-664112216)

choyrim, 2020-07-27:
@nick-stocks so to be clear, you are good now?

nick-stocks, 2020-07-28:
@choyrim Yes, all good thanks!

choyrim, 2020-07-29:
Apparently @kageurufu already submitted PR #381 to do the CAP_V4L hack mentioned above. Unfortunately it fails the CI build.

choyrim, 2020-07-29:
and before that @ajnart seems to have merged a fix for the bug #318 in PR #319 ... but put the CAP_V4L in the cli scripts. however i don't see the fix in the master branch. Seems like the history was changed. Looking at the history of cli/test.py with the PR
https://github.com/boltgolt/howdy/commits/9d35ccfee95e4cd426050ce83abf8b82ab588f98/src/cli/test.py
vs the history of cli/test.py in master, it looks like the history was replaced/rebased/whatever
https://github.com/boltgolt/howdy/commits/master/src/cli/test.py

so it seems the fix was lost somewhere.

miskovics, 2020-08-02:
Just want to report that I've just installed Howdy on Ubuntu 20.04 and I encountered the same issue: first 2 attempts failed

```
Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 66, in <module>
    cv2.setMouseCallback("Howdy Test", mouse)
cv2.error: OpenCV(4.3.0) /io/opencv/modules/highgui/src/window_QT.cpp:717: error: (-27:Null pointer) NULL window handler in function 'cvSetMouseCallback'
```

the 3rd worked, showing an empty window and printing this on console:

```
...
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x8e0000f
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x8e0000f
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x8e0000f
^C
Closing window
```

yimisunrise, 2020-08-03:
@nick-stocks 
thanks ! This approach solves my problem。

boltgolt, 2020-08-10:
Fix looks great, but the PR #381 makes the automated tests fail to read the camera. Original issue is fixed though so closing thread

closed 2020-08-10 by boltgolt