boltgolt/howdy #393: Not working on Dell Lattitude 3500

archit2604, 2020-07-14:
_Please describe the issue in as much detail as possible, including any errors and traces_
ImportError: numpy.core.multiarray failed to import
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 17, in <module>
    import cv2
  File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 5, in <module>
    from .cv2 import *
ImportError: numpy.core.multiarray failed to import
Unknown error: 1

I have tried reinstalling and updating numpy but it doesn't work

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution:Ubuntu 18.04

Howdy version:Latest

hanaral, 2020-07-14:
I suspect it's to do with the opencv package in 18.04 being quite out of date, since I got a similar error trying to build an opencv project that required an up to date version.

archit2604, 2020-07-15:
How to fix?