boltgolt/howdy #397: Should snapshots be readable for everyone?

Jorgen-P, 2020-07-16:
I had a look around for the snapshot folder and discovered that the file in it was readable for everyone, I also tried opening it in an image viewing program and that worked. The permissions on it are:
-rw-rw-r-- root <my user's group> /usr/lib/security/howdy/snapshots/20200716T180150.jpg

OS: Kubuntu 20.04
Howdy version: 2.6.0

boltgolt, 2020-08-09:
That would be a lot better, yes. It might make it harder for people sending snapshots for debug purposes

saychakra, 2020-09-01:
I modified the snapshot.py code, to only store the data of the latest snapshot and remove all existing snapshots. The main reason for this was mainly to limit the size of the snapshot folder, which was getting absolutely huge! Everything is working fine as of now. Don't know whether there would be any issues after any update.

arifer612, 2020-12-02:
I have a question regarding the snapshots folder. Is it necessary to keep so many pictures if there is already a user model in the `models` folder?

patrakov, 2021-09-10:
This is a security issue. An attacker (another legitimate user of this PC) could trivially use the photos taken during successful authentication of the target user in order to impersonate the target user. I think this at least warrants a comment.

boltgolt, 2021-09-11:
That's a good point, but "trivially easy" is a severe overstatement. Feeding the images back into Howdy requires either the code or the config file to be changed so a different (virtual/loopback) camera device can be configured. All these files should only be editable by root. If the root user is compromised then why even mess with Howdy? 

The other option would be cloning the USB ID of the camera and using a dedicated board to impersonate it. This is only an option when you have physical access to the device and the camera is not build into a laptop. 

You raise a good point but exaggerate the impact a bit. On 3.0.0 both options are disabled by default, as they should have been in the first place.

patrakov, 2021-09-11:
There are lots of instructions (mostly related to making howdy work with screensavers) that ask for changing the permissions recursively. So de-facto, if a person has successfully set up howdy to work with the screensaver, root is not needed for accessing the snapshots. And feeding back images to Howdy via the pre-existing camera requires only a printer and some paper, no need to mess with virtual cameras or custom USB devices.

So no, while this indeed does not apply to the new default configuration and default permissions, this is not an overstatement. Anyway, I understand that the next version will have the default changed - but could you please add some comments regarding this concern to the default config file?

boltgolt, 2021-09-12:
I mean, yeah. If you modify a program like Howdy to be explicitly less secure i don't think you should be shocked to find it easier to bypass. This is only needed for some specific screensavers because they have a non-compliant PAM implementation. And while printing remains the easiest attack, you need a printer that prints in the IR spectrum, [which not all printers do](https://docs.microsoft.com/en-us/windows-hardware/design/device-experiences/windows-hello-face-authentication#benefits-of-near-infrared). Even then, you're just as vulnerable if a photo of you has ever been posted online.

So many of the default config options harm the security of Howdy: `abort_if_ssh`, `abort_if_lid_closed`, `certainty` and `device_path` are the most dangerous. I guess it would mean adding warnings to all of them. I'll think about it.