boltgolt/howdy #398: Camera path is not configured correctly, please edit the 'device_path' config value

reylinux, 2020-07-17:
Hi, I got stuck on adding my face. I am still very new to Linux and not sure how to solve this myself.

Can anyone help me out?

Adding face model for the user reynaldi
Enter a label for this new model [Initial model] (max 24 characters): Rey
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <bound method VideoCapture.__del__ of <recorders.video_capture.VideoCapture object at 0x7ff16836f438>>
Traceback (most recent call last):
  File "/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'

Linux distribution (if applicable):
Elementary OS Hera

Howdy version:
2.6.0

pandom79, 2020-07-17:
run 

`sudo howdy config`

find **device_path** and replace its value with **/dev/video0** or **/dev/video1** etc etc
One some values it will has to works.
After that, run 

`sudo howdy test 
`
for testing it

reylinux, 2020-07-17:
Awesome! It seems to be working with the sudo howdy test.

I'll let you know if it work soon

reylinux, 2020-07-17:
> run
> 
> `sudo howdy config`
> 
> find **device_path** and replace its value with **/dev/video0** or **/dev/video1** etc etc
> One some values it will has to works.
> After that, run
> 
> `sudo howdy test `
> for testing it

Hi Phandom, I have a new issue now:

~$ sudo howdy test
Face detection timeout reached

This is what I got and I couldn't login to my account before as well

pandom79, 2020-07-17:
**sudo howdy test** command is only for testing if camera works. 
Now, you have to add a face model for user.
run
`sudo howdy add`
you add at least 2 face model...so you are more secure that howdy works.

reylinux, 2020-07-17:
> **sudo howdy test** command is only for testing if camera works.
> Now, you have to add a face model for user.
> run
> `sudo howdy add`
> you add at least 2 face model...so you are more secure that howdy works.
> 
> forgot say make a reboot could help you

Thank you!! Now it is working (I added 4 models earlier with the different lighting situations.

closed 2020-07-18 by reylinux