boltgolt/howdy #400: Unable unlock randomally

pandom79, 2020-07-18:
Unable unlock randomally. 
In the logs i only read :
`Attempting facial authentication for user MYUSER `
The camera is on and it remains on. 
I can't read 
`login approved`
I have to a hard reboot.
There aren't errors in logs.
It has a random behaviour.
When there aren't errors is the worse way for to diagnostic a bug.
What happens?

boltgolt, 2020-08-07:
What version of Howdy are you on (`sudo howdy version`) and what distro?

pandom79, 2020-08-07:
> What version of Howdy are you on (`sudo howdy version`) and what distro?

Howdy 2.6.0
Void Linux

It stucks randomally on this line in pam.py

```
# Run compare as python3 subprocess to circumvent python version and import issues
status = subprocess.call(["/usr/bin/python3", os.path.dirname(os.path.abspath(__file__)) + "/compare.py", pamh.get_user()])
```

If this you can help

boltgolt, 2020-08-10:
That line is basically what executes the entire program (calling the core script) so no surprises there. Does it happen in different lighting environments?

pandom79, 2020-08-10:
> That line is basically what executes the entire program (calling the core script) so no surprises there. Does it happen in different lighting environments?

Before to insert howdy authentication in important  pam file , i tried to put it in sudo file (/etc/pam.d/sudo) for testing it.
Initially it works fine, then 
Suddenly, after sudo command,  it remains blocked with led camera on and in log file i see "[HOWDY] Attempting facial authentication for user". I need to kill process to unblock it.

boltgolt, 2020-08-11:
Are there any errors in `/var/log/apache2/error.log`?

pandom79, 2020-08-11:
I see in log file only following lines without errors:
```
2020-08-11T16:02:14.70687 auth.emerg: Aug 11 18:02:14 sudo: [HOWDY] Attempting facial authentication for user MYUSER
```
I haven't apache installed.
When there aren't errors , it more difficult to find a workaround

boltgolt, 2020-09-28:
Sorry, i mentioned the wrong log file, should have been `/var/log/auth.log`

pandom79, 2020-10-03:
Perhaps i found the problem. I added howdy in the **/etc/pam.d/other** file, for the screen locker (KDE).
I think that it is wrong. I moved that in **/etc/pan.d/kde** and it seems works fine.
Anyway, I'll test it for some days, if it works fine then I'll close this issue.

pandom79, 2020-10-03:
I spoke too soon. Suddenly it stops with the LED camera on. On archlinux it works fine. I checked all packages and related versions. I don't understand because itself blocks

pandom79, 2020-10-07:
I added howdy in the following files:
**/etc/pam.d/sudo**
**/etc/pam.d/sddm  (login)**
**/etc/pam.d/kde      (screenlocker)**

It works fine but suddenly i get this error:

```
(python3:5308): GStreamer-CRITICAL **: 14:25:43.249: 
Trying to dispose element pipeline0, but it is in READY instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.
```

closed 2026-11-16 by boltgolt