boltgolt/howdy #404: Add operation crashes

cnyk, 2020-07-28:
On Fedora 32 howdy add operation crashes with "Floating point exception".
By moving the import cv2 call after the import dlib call, the crash does not accur
Patch here: https://github.com/cnyk/howdy/commit/40f9e31e8106b6178f5dc049697b87179d664fe0

boltgolt, 2020-08-07:
Thanks for figuring it out! Could you open a pull request for that patch so i can attribute it to you properly?

tomoshokas, 2020-08-07:
Core was generated by `/usr/bin/python3 /usr/bin/howdy add'.
Program terminated with signal SIGFPE, Arithmetic exception.
#0 0x00007f82bcb5588f in sgemm_thread_tn (args=0x7fff37af4c50, range_m=0x0, range_n=0x0, sa=0x7f8296000000, sb=0x7f8296064000, mypos=) at level3_thread.c:759
759 nthreads_n = (n + SWITCH_RATIO * nthreads_m - 1) / (SWITCH_RATIO * nthreads_m);
[Current thread is 1 (Thread 0x7f82da9d0740 (LWP 31604))]

The formula comes from /usr/lib64/libopenblas.so.0 vesrion 0.3.9-3, no downgrade possible through dnf.

Re-installation of OpenCV and python3-opencv does not help.

closed 2020-09-02 by boltgolt