boltgolt/howdy #407: Not working Thinkpad E15 Gen2 (AMD)

xpusostomos, 2020-07-30:
I've got the new model Thinkpad E15 Gen 2 here, and it doesn't work. There was some message during installation, that unfortunately scrolled off the screen about not being able to recognise a video device and that "some manual configuration may be required". When I run sudo howdy add I get:
```
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7f26c0c3c550>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'
```

The output of lsusb for what I presume is the video device is:

```

Bus 001 Device 002: ID 13d3:56fb IMC Networks Integrated Camera
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x13d3 IMC Networks
  idProduct          0x56fb 
  bcdDevice           20.01
  iManufacturer           3 Azurewave
  iProduct                1 Integrated Camera
  iSerial                 2 0000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0532
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 Integrated Camera
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               5 Integrated Camera
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              5 Integrated Camera
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.10
        wTotalLength       0x0087
        dwClockFrequency       15.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
[... 937 lines cut ...]
        wTransferSize                    4096 bytes
        bcdDFUVersion                   1.10
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0029
  bNumDeviceCaps          2
  Platform Device Capability:
    bLength                28
    bDescriptorType        16
    bDevCapabilityType      5
    bReserved               0
    PlatformCapabilityUUID    {d8dd60df-4589-4cc7-9cd2-659d9e648a9f}
    CapabilityData[0]    0x00
    CapabilityData[1]    0x00
    CapabilityData[2]    0x03
    CapabilityData[3]    0x06
    CapabilityData[4]    0xe0
    CapabilityData[5]    0x04
    CapabilityData[6]    0x15
    CapabilityData[7]    0x00
  ** UNRECOGNIZED:  08 10 11 01 03 00 00 00
Device Status:     0x0000
  (Bus Powered)
```

xpusostomos, 2020-07-30:
I changed howdy config to add what I presume ought to be the right device:

```
device_path = /dev/v4l/by-path/pci-0000:05:00.3-usb-0:3:1.0-video-index0
```

However, no real success. Howdy test does this:

```
Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 66, in <module>
    cv2.setMouseCallback("Howdy Test", mouse)
cv2.error: OpenCV(4.3.0) /io/opencv/modules/highgui/src/window_QT.cpp:717: error: (-27:Null pointer) NULL window handler in function 'cvSetMouseCallback'
```

boltgolt, 2020-08-07:
Looks like your video device can't be read correctly, can you view your camera feed in a program like cheese?

xpusostomos, 2020-08-07:
Yes, I can see myself when using Cheese.

boltgolt, 2020-08-07:
Do you see yourself in black and white?

xpusostomos, 2020-08-07:
I can see colours.

boltgolt, 2020-08-10:
Then you're using the normal webcam, try switching to the IR camera in the options

xpusostomos, 2020-08-11:
How would I switch to the IR camera? I tried all the other device paths in that directory of devices, but no cigar.

Also oddly, "howdy add" seems to now not complain when I attempt to use it. But "howdy test" comes up with a flat grey window and spews this out continually on the console:

X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x520000f

Mmmm, now running it again, no changes to the config, it puts out the  error I reported up the top.

boltgolt, 2020-08-11:
Check out #60

xpusostomos, 2020-08-14:
I tried the QT_X11_NO_MITSHM=1 thing, and as far as I can tell it has no effect on behavior, HOWEVER, with all the playing around I found howdy test actually comes up with a window about 1 in 5 times. None of the documentation says what I'm supposed to see at this point, but it looks like it's working. I see myself in black and white, I see a red circle around my face, I see green numbers in the lower left.

However, while it works about 1 in 5 times, the other times it crashes with the error above about /window_QT.cpp:717.  As far as I can tell QT_X11_NO_MITSHM=1 has no effect on its frequency of success. I also tried the force_mpeg which seems to have no effect, and I tried upping the certainty to 5

I also often get this when I click on the test image:
```
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 186, in <module>
    time.sleep(.5 - frame_time)
ValueError: sleep length must be non-negative
```
Having said all that, it does actually seem to be working for me now to sudo and to login. I'm presuming upping the certainty number made the difference.

xpusostomos, 2020-08-14:
I stand corrected, the QT_X11_NO_MITSHM=1 is necessary for it to work. But it doesn't stop howdy test crashing 4 out of 5 times.

Hnagzhi, 2020-08-15:
So I've got the E14 Gen2 with the same webcam and using /dev/video0 lets howdy use the normal webcam and /dev/video2 for the IR one. However, there is still no way of turning the IR blaster on (only on chicony based cams afaik)...

xpusostomos, 2020-08-15:
If I use /dev/video2 for howdy test, the quality is awful. But if I use  /dev/v4l/by-path/pci-0000:05:00.3-usb-0:3:1.0-video-index0 the quality decent.  What this means, I don't know.

xpusostomos, 2020-08-25:
Playing around in Windows, a reddish light comes on during face detection, and I've never seen that under Linux, so.... it doesn't seem like I've stumbled across anything to make the IR work.

hannesaronsson, 2020-08-25:
> Playing around in Windows, a reddish light comes on during face detection, and I've never seen that under Linux, so.... it doesn't seem like I've stumbled across anything to make the IR work.

I have this exact issue as well, on the Lenovo Slim 7. When using the IR-camera (/dev/video2) the IR-leds are not flashing. I wonder if there is any simple fix to this issue.

ONDRA5, 2020-08-31:
> So I've got the E14 Gen2 with the same webcam and using /dev/video0 lets howdy use the normal webcam and /dev/video2 for the IR one. However, there is still no way of turning the IR blaster on (only on chicony based cams afaik)...

I'm on Thinkpad E14 gen2 and I have the chicony one and so far I wasn't able to get the IR blaster working.. if there's a way, could you please point me in it's direction? Thanks

xpusostomos, 2020-08-31:
@ONDRA5 ... I couldn't get IR working, but I could get howdy working without it.

cedjo, 2020-09-03:
added device path in config:
device_path = /dev/v4l/by-path/pci-0000:04:00.3-usb-0:3:1.0-video-index0
howdy test shows a great gray image!

howdy add fails with this errors:

[XXXX@thinkpad-e15 by-path]$ sudo howdy --user XXXX add
Adding face model for the user XXXX
Enter a label for this new model [Initial model] (max 24 characters): face1
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/videoio_registry.cpp (187) VideoBackendRegistry VIDEOIO: Enabled backends(7, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for FFMPEG
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (690) open OpenCV | GStreamer: /dev/v4l/by-path/pci-0000:04:00.3-usb-0:3:1.0-video-index0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (723) open OpenCV | GStreamer: mode - FILE
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (1758) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (342) getPluginCandidates VideoIO pluigin (INTEL_MFX): glob is 'libopencv_videoio_intel_mfx*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (349) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.2.0/modules/videoio/src/backend_plugin.cpp (353) getPluginCandidates Found 0 plugin(s) for INTEL_MFX

Please look straight into the camera
Floating point exception

and stops...
Running on Fedora 32

bobzhu418, 2020-09-06:
> I have this exact issue as well, on the Lenovo Slim 7. When using the IR-camera (/dev/video2) the IR-leds are not flashing. I wonder if there is any simple fix to this issue.

+1

filip-hejsek, 2021-05-07:
The right camera to use on Thinkpad E15 Gen2 is **/dev/video2**, which is the IR camera. When you open the camera you should see a black and white image.
For the IR camera to work, you need to enable the IR emmiter, see #269, TL;DR you need to install https://github.com/EmixamPP/linux-enable-ir-emitter