boltgolt/howdy #408: unknown error 1

orclex, 2020-07-31:
I have the same problem as @HZ89 in

_Originally posted by @boltgolt in https://github.com/boltgolt/howdy/issues/243#issuecomment-648098342_

and I didn't find a new ticket for this problem as suggested by @boltgolt. So here it is...

howdy is working in general. "sudo su" with authentication through howdy works. Screen unlocking (KDE Plasma) does not work and shows "unknown error 1".
Setting permissions
  chmod o+x /lib/security/howdy/dlib-data
does not fix this. It does not work since (K)ubuntu 20.04 and howdy 2.6.0 where I also configured fingerprint as additional pam authentication module if this could be a problem.
Is there any solution?

boltgolt, 2020-08-01:
Does anything show up in `tail /var/log/auth.log` or when using sudo in the terminal?

amcolash, 2020-08-02:
Same here! A week or two ago just magically stopped working.

Here is a bit of one of my logs:
```
Aug  2 11:18:14 andrew-laptop xfce4-screensaver-dialog: [HOWDY] Attempting facial authentication for user amcolash
Aug  2 11:18:16 andrew-laptop /lib/security/howdy/pam.py[164224]: Traceback (most recent call last):
Aug  2 11:18:16 andrew-laptop /lib/security/howdy/pam.py[164224]:   File "/lib/security/howdy/pam.py", line 84, in pam_sm_authenticate
Aug  2 11:18:16 andrew-laptop /lib/security/howdy/pam.py[164224]:     return doAuth(pamh)
Aug  2 11:18:16 andrew-laptop /lib/security/howdy/pam.py[164224]:   File "/lib/security/howdy/pam.py", line 77, in doAuth
Aug  2 11:18:16 andrew-laptop /lib/security/howdy/pam.py[164224]:     pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Unknown error: " + str(status)))
Aug  2 11:18:16 andrew-laptop /lib/security/howdy/pam.py[164224]: PamException: Conversation error
```

orclex, 2020-08-04:
This is what I get in /var/log/auth.log if trying to authenticate with user alex in KDE sddm loginmanager:
 

```
Aug  4 09:00:50 m19040-lin kcheckpass: pam_sss(kde:auth): authentication failure; logname= uid=1001 euid=1001 tty=:0 ruser= rhost= user=alex
Aug  4 09:00:50 m19040-lin kcheckpass: pam_sss(kde:auth): received for user alex: 10 (User not known to the underlying authentication module)
Aug  4 09:00:53 m19040-lin kcheckpass: [HOWDY] Attempting facial authentication for user alex
Aug  4 09:00:53 m19040-lin kcheckpass: [HOWDY] Failure, unknown error1
Aug  4 09:00:53 m19040-lin unix_chkpwd[33195]: password check failed for user (alex)
Aug  4 09:00:53 m19040-lin kcheckpass: pam_unix(kde:auth): authentication failure; logname= uid=1001 euid=1001 tty=:0 ruser= rhost=  user=alex
Aug  4 09:00:53 m19040-lin kcheckpass: pam_sss(kde:auth): authentication failure; logname= uid=1001 euid=1001 tty=:0 ruser= rhost= user=alex
Aug  4 09:00:53 m19040-lin kcheckpass: pam_sss(kde:auth): received for user alex: 10 (User not known to the underlying authentication module)
Aug  4 09:00:59 m19040-lin kcheckpass: [HOWDY] Attempting facial authentication for user alex
Aug  4 09:01:00 m19040-lin kcheckpass: [HOWDY] Failure, unknown error1
```

Same user alex with sudo on a terminal session is working:

```
Aug  4 09:01:41 m19040-lin sudo: [HOWDY] Login approved
Aug  4 09:01:41 m19040-lin sudo:     alex : TTY=pts/5 ; PWD=/home/alex ; USER=root ; COMMAND=/usr/bin/su
Aug  4 09:01:41 m19040-lin sudo: pam_unix(sudo:session): session opened for user root by alex(uid=0)
Aug  4 09:01:41 m19040-lin su: (to root) alex on pts/5
Aug  4 09:01:41 m19040-lin su: pam_unix(su:session): session opened for user root by alex(uid=0)
```

howdy config is:
```
[core]
#detection_notice = false
detection_notice = true
#no_confirmation = false
no_confirmation = true
suppress_unknown = false
ignore_ssh = true
ignore_closed_lid = true
disabled = false
use_cnn = false

[video]
certainty = 3.5
timeout = 4
device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:8:1.2-video-index0
max_height = 320
frame_width = -1
frame_height = -1
dark_threshold = 100
recording_plugin = opencv
device_format = v4l2
force_mjpeg = false
exposure = -1

[snapshots]
capture_failed = true
capture_successful = true

[debug]
end_report = false
```

Kubuntu 20.04, howdy 2.6.0

boltgolt, 2020-08-07:
Very weird @amcolash, but it's a dependency issue. Maybe reinstalling pam-python will help?

---

@orclex have you tried setting everything in /lib/security/howdy to world readable?

orclex, 2020-08-07:
Yes:
# chmod -R a+r /lib/security/howdy

# ll /lib/security/howdy/
insgesamt 32
drwxr-xr-x 1 root root  216 Aug  7 11:26 ./
drwxr-xr-x 1 root root   40 Mai 12 13:38 ../
drwxr-xr-x 1 root root  170 Jun 25 10:08 cli/
-rwxr-xr-x 1 root root 3082 Jun 22 19:19 cli.py*
-rw-r--r-- 1 root root 9272 Jun 22 19:19 compare.py
-rw-r--r-- 1 root root  560 Aug  7 11:26 config.ini
drwxr--r-x 1 root root  248 Jun 25 10:08 dlib-data/
-rw-r--r-- 1 root root 3068 Jun 22 19:25 logo.png
drwxr-xr-x 1 root root   16 Mai 12 13:59 models/
drwxr--r-- 1 root root   10 Jun 25 10:08 pam-config/
-rw-r--r-- 1 root root 3554 Jun 22 19:19 pam.py
drwxr-xr-x 1 root root   46 Jun 23 08:21 __pycache__/
drwxr--r-- 1 root root  154 Jun 25 10:08 recorders/
-rw-r--r-- 1 root root 1835 Jun 22 19:19 snapshot.py
drwxr-xr-x 1 root root 3154 Aug  7 11:27 snapshots/

Unlocking lockscreen in KDE with howdy is not possible anymore which was working before. I get the error "Unlocking failed" immediately after pressing return in the login screen. It does not switch on camera and IR. If I enter the password I can login but I see for a very short moment "unknown error: 1".
sudo is still working with howdy. 

  

Aug  7 11:38:58 m19040-lin kcheckpass: [HOWDY] Attempting facial authentication for user alex
Aug  7 11:38:59 m19040-lin kcheckpass: [HOWDY] Failure, unknown error1
Aug  7 11:38:59 m19040-lin unix_chkpwd[6608]: password check failed for user (alex)
Aug  7 11:38:59 m19040-lin kcheckpass: pam_unix(kde:auth): authentication failure; logname= uid=1001 euid=1001 tty=:0 ruser= rhost=  user=alex
Aug  7 11:38:59 m19040-lin kcheckpass: pam_sss(kde:auth): authentication failure; logname= uid=1001 euid=1001 tty=:0 ruser= rhost= user=alex
Aug  7 11:38:59 m19040-lin kcheckpass: pam_sss(kde:auth): received for user alex: 10 (User not known to the underlying authentication module)

amcolash, 2020-08-09:
No clue what changed, but did an upgrade of all apt packages this weekend and now things are back to normal? Will let you know if things are not working again. Thanks for the help!

orclex, 2020-08-10:
Unfortunately no change here on Kubuntu 20.04.. :-(

boltgolt, 2020-08-10:
If it can be fixed by upgrading other stuff a reinstall might fix it too

orclex, 2020-08-13:
Unfortunately this does not work for me.
I did:

```
apt purge howdy
apt autoremove

apt install howdy
pip install dlib
```

and added my face again. sudo is working as before but login via sddm on kubuntu gives me an "unknown error" without switching on the ir and camera LEDs.

CChiappa, 2020-08-20:
I had the same problem with a new install of Howdy on Debian.  I fiddled pam.py to strace the subcall to compare.py and noticed a directory which was missing execute privs for non-root. This fixed it:
sudo chmod +x /lib/security/howdy/recorders/
This may be a Debian packaging issue?

CChiappa, 2020-08-20:
I should say - the plugin seems to work now, I get a message about "recognized face Chris" or similar, but the pam integration is still broken - the screensaver still wants my password.  I won't have time to look at that for a bit.

CChiappa, 2020-08-30:
When I set no_confirmation = true as mentioned in 
https://github.com/boltgolt/howdy/issues/65 everything seems to work!

rmacordeiro, 2020-09-01:
Hi,

I'm getting this same error "unknown error 1" when trying to unlock after screensaver.
It works just fine on the terminal but when unlocking when on the screensaver it shows just this on the log file:
_Sep  1 15:11:43 rcordeiro-MacBookPro xscreensaver: [HOWDY] Attempting facial authentication for user rcordeiro
Sep  1 15:11:51 rcordeiro-MacBookPro xscreensaver: [HOWDY] Failure, unknown error1_

On the screen I get this:
_Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 95, in <module>
     models = json.load(open(PATH + "/models/" + user + ".dat"))
PermissionError: [Errno 13] Permission denied: '/lib/security/howdy/models/rcordeiro.dat'_

I have added full permissions on the mentioned file, but still no luck.

I'm using Ubuntu 18.04.
My screensaver is Xscreensaver.

Any ideas?

Thank you

mandelbroo, 2020-10-03:
Had same the same issue on Linux Mint 20 with screen-saver. Running `sudo apt install python3-opencv` has fixed the problem.
Thanks @rmacordeiro! And thank you @boltgolt and contributors for a great piece of software!

cccaballero, 2021-07-02:
Hello, similar issue here, but using Ubuntu 20.04, after a recent system update I get the `unknown error 1` message only when I try to start session, after that it works well, even if y close the session and try again.

dominalien, 2021-08-28:
Hello, I'm having the same Unknown error: 1 when trying to unlock the screen in Tuxedo OS, which is basically Ubuntu Budgie. I can log in and sudo -i, but when unlocking the screen I immediately get the error. I did both things in "Lockscreen not unlocking" on the wiki, no effect. I installed python3-opencv, no effect.

/var/log/auth.log tells me:
Aug 28 21:02:27 domin-TUXEDO-InfinityBook-Pro-14-Gen6 [HOWDY]: Attempting facial
 authentication for user domin
Aug 28 21:02:27 domin-TUXEDO-InfinityBook-Pro-14-Gen6 /lib/security/howdy/pam.py
[16455]: Traceback (most recent call last):
Aug 28 21:02:27 domin-TUXEDO-InfinityBook-Pro-14-Gen6 /lib/security/howdy/pam.py
[16455]:   File "/lib/security/howdy/pam.py", line 93, in pam_sm_authenticate
Aug 28 21:02:27 domin-TUXEDO-InfinityBook-Pro-14-Gen6 /lib/security/howdy/pam.py
[16455]:     return doAuth(pamh)
Aug 28 21:02:27 domin-TUXEDO-InfinityBook-Pro-14-Gen6 /lib/security/howdy/pam.py
[16455]:   File "/lib/security/howdy/pam.py", line 85, in doAuth
Aug 28 21:02:27 domin-TUXEDO-InfinityBook-Pro-14-Gen6 /lib/security/howdy/pam.py
[16455]:     pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Unknown error: 
" + str(status)))
Aug 28 21:02:27 domin-TUXEDO-InfinityBook-Pro-14-Gen6 /lib/security/howdy/pam.py
[16455]: PamException: Conversation error
Aug 28 21:02:27 domin-TUXEDO-InfinityBook-Pro-14-Gen6 gnome-screensaver-dialog[1
6455]: pam_unix(gnome-screensaver:auth): Couldn't open /etc/securetty: No such f
ile or directory
Aug 28 21:02:27 domin-TUXEDO-InfinityBook-Pro-14-Gen6 gnome-screensaver-dialog[1
6471]: pam_unix(gnome-screensaver:auth): Couldn't open /etc/securetty: No such f
ile or directory
Aug 28 21:02:32 domin-TUXEDO-InfinityBook-Pro-14-Gen6 gnome-screensaver-dialog[1
6473]: pam_unix(gnome-screensaver:auth): Couldn't open /etc/securetty: No such f
ile or directory
Aug 28 21:02:32 domin-TUXEDO-InfinityBook-Pro-14-Gen6 gnome-screensaver-dialog[1
6455]: gkr-pam: unlocked login keyring

boltgolt, 2021-09-02:
@dominalien seems like a PAM issue there, could you make sure that `detection_notice = false` and `no_confirmation = true` in the config?

dominalien, 2021-09-02:
@boltgolt Yes, both are set exactly like that. I double checked restarted and still getting Unknown error: 1 on unlock screen. Everything else works fine.

jeroen1243, 2021-10-02:
Same issue here.
`Oct  2 15:57:47 pop-os [HOWDY]: Failure, unknown error1`
Only on first startup/reboot i get this error message. 

And also only the first time, because when i type a wrong password and facial recognition is run again it doesn't give an error and unlocks the device.